natario1/CameraView #416: Can't update to V2

ManuEpi, 2019-03-21:
### How do I?

Hello , I'm trying to update to V2 but unfortunately I can't , the problem is this line : 
     `cameraView.setLifecycleOwner((LifecycleOwner) this);`
I don't have AndroidX LifecycleOwner library , and if I add it , I have this message : 
`Manifest merger failed : Attribute application@appComponentFactory value=(android.support.v4.app.CoreComponentFactory) from [com.android.support:support-compat:28.0.0] AndroidManifest.xml:22:18-91
	is also present at [androidx.core:core:1.0.0] AndroidManifest.xml:22:18-86 value=(androidx.core.app.CoreComponentFactory).
	Suggestion: add 'tools:replace="android:appComponentFactory"' to <application> element at AndroidManifest.xml:14:5-40:19 to override.
`

What can I do ?

### Version used
CameraView Version 1.6.1

cedricdevriendt, 2019-03-21:
Try migrating the entire project to AndroidX since you now have a combination of old (support library) and new (AndroidX) : https://developer.android.com/jetpack/androidx/migrate#migrate

ManuEpi, 2019-03-21:
I wanted to not migrate my entire project , I'm afraid the migration is going badly...

ManuEpi, 2019-03-21:
Whatever , I migrated all my project but that didn't resolve my problem which is Media recorder failed to start when I'm trying to take a video facing back ..

closed 2019-03-21 by ManuEpi