natario1/CameraView #417: Camera service died

ManuEpi, 2019-03-21:
### How do I?
Device : Not quite a phone , it's an "Rockchip rk3288" API 23

I want to take picture/video with camera Facing BACK

My problem here is that on my device , I'm able to take every picture I want, but when I try to take video , it shows me this error every time , so the recording fails :

`W/CameraBase: Camera service died!
W/IMediaDeathNotifier: media server died
W/CameraBase: mediaserver's remote binder Camera object died
E/Camera: Error 100
I/CameraController: Start: posting runnable. State: STATE_STARTED
W/Camera1: Recoverable error inside the onError callback. CAMERA_ERROR_SERVER_DIED
I/CameraController: stopImmediately: State was: STATE_STARTED
I/Camera1: onStop: About to clean up.
I/CameraCallbacks: dispatchOnVideoTaken com.otaliastudios.cameraview.VideoResult@2993d38
D/Camera: app passed NULL surface
I/Camera1: destroyCamera: Clean up. Releasing camera.
    destroyCamera: Clean up. Released camera.
W/Camera1: onStop: Clean up. Returning.
I/CameraController: stopImmediately: Stopped. State is: STATE_STOPPED
I/CameraController: Start: posting runnable. State: STATE_STOPPED
    Start: executing. State: STATE_STOPPED
    Start: about to call onStart() STATE_STARTING
I/Camera1: collectCameraId Facing: BACK Internal: 0 Cameras: 1
I/Camera1: createCamera: Applying default parameters.
I/Camera1: bindToSurface: Started
I/CameraController: computeCaptureSize: result: 2592x1944 flip: false mode: VIDEO
I/Camera1: size: sizesFromList: [2592x1944, 1920x1080, 640x480, 1280x720, 1280x1024]
I/CameraController: size: computePreviewStreamSize: targetRatio: 4:3 targetMinSize: 1080x1788
I/CameraController: computePreviewStreamSize: result: 2592x1944 flip: false
I/Camera1: onStart Dispatching onCameraPreviewStreamSizeChanged.
I/CameraCallbacks: onCameraPreviewStreamSizeChanged
I/CameraPreview: setStreamSize: desiredW= 2592 desiredH= 1944
I/CameraController: computeCaptureSize: result: 2592x1944 flip: false mode: PICTURE
I/Camera1: onStart Starting preview with startPreview().
I/Camera1: onStart Started preview.
    onStart: Ended
I/CameraController: Start: returned from onStart(). Dispatching. STATE_STARTING
I/CameraCallbacks: dispatchOnCameraOpened com.otaliastudios.cameraview.CameraOptions@66adde5
I/CameraController: Start: executing. State: STATE_STARTED
I/CameraView: onMeasure: requested dimensions are (1080[EXACTLY]x1788[EXACTLY])
    onMeasure: previewSize is (2592.0x1944.0)
W/CameraView: onMeasure: both are MATCH_PARENT or fixed value. We adapt. This means CROP_CENTER. (1080x1788)
`

But the same codes , works on an Huawei API 23 , and Galaxy J7 api 27
I'm able to take picture/videos on both with camera facing BACK too.

Do you have any idea guys ?

Thanks

### Version used
CameraView version 2.0.0-beta 4 ( latest )

ManuEpi, 2019-03-22:
And with cameraView Version 1.6.1 I have this error message : 
E/MediaRecorder: start failed: -19
E/Camera1: Error while starting MediaRecorder. Swallowing. java.lang.RuntimeException: start failed.
    java.lang.RuntimeException: start failed.
        at android.media.MediaRecorder.start(Native Method)
        at com.otaliastudios.cameraview.Camera1$14.run(Camera1.java:645)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:63)
        at android.os.Handler.handleCallback(Handler.java:743)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:148)
        at android.os.HandlerThread.run(HandlerThread.java:61)
I/Camera1: endVideoImmediately: is capturing: true
E/MediaRecorder: stop called in an invalid state: 0
W/Camera1: endVideoImmediately: Error while closing media recorder. Swallowing java.lang.IllegalStateException
    java.lang.IllegalStateException
        at android.media.MediaRecorder.stop(Native Method)
        at com.otaliastudios.cameraview.Camera1.endVideoImmediately(Camera1.java:675)
        at com.otaliastudios.cameraview.Camera1.access$1500(Camera1.java:27)
        at com.otaliastudios.cameraview.Camera1$14.run(Camera1.java:650)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:63)
        at android.os.Handler.handleCallback(Handler.java:743)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:148)
        at android.os.HandlerThread.run(HandlerThread.java:61)

natario1, 2019-03-23:
I have no idea, sorry! Each device is different (e.g. about video codecs and their quality) so I can only tell you to try to experiment yourself with our source code.

ManuEpi, 2019-03-25:
Yeah no problem , I think that the problem is the device , because it's not stable .
Actually this device with the original camera can take video in 720P , but when I switch to 480P , it crashs...
I tried to take video with your library in 720P to make it works but it doesn't
Seems mysterious

closed 2019-04-21 by stale[bot]

arturschaefer-hotmart, 2020-05-11:
The problem persists

arturschaefer-hotmart, 2020-05-11:
Please, re-open this issue. I have the same problem using the Moto X.

ManuEpi, 2020-05-11:
The problem for me was my device , the external camera didn't support 480p recording ( which is set by default by the library ) I think . So I changed video quality option, to record with the max quality (1080p) and it worked 
The problem here was totally my device

arturschaefer-hotmart, 2020-05-11:
@ManuEpi , thanks for the help. How did you do this? I got the legacy code with this lib and don't know exactly how to fix.

ManuEpi, 2020-05-11:
I was using an older version (my problem was on March 2019) . And I set the value app:cameraVideoQuality to "highest" . But I don't find it in the current documentation :/ it may have been deleted ?

arturschaefer-hotmart, 2020-05-11:
Ok... thanks for your help. It was really helpful.