natario1/CameraView #418: Ready for Android Q (CAMERA permission)?

hardysim, 2019-03-22:
See https://developer.android.com/preview/privacy/camera-connectivity#camera-info-permission

Apps running on Android Q using `getCameraCharacteristics()` will need the `CAMERA` permission to access some information in it.

> If your app doesn't have the CAMERA permission, it cannot access the following fields:
> 
> - ANDROID_LENS_POSE_ROTATION
> - ANDROID_LENS_POSE_TRANSLATION
> - ANDROID_LENS_INTRINSIC_CALIBRATION
> - ANDROID_LENS_RADIAL_DISTORTION
> - ANDROID_LENS_POSE_REFERENCE
> - ANDROID_LENS_DISTORTION
> - ANDROID_LENS_INFO_HYPERFOCAL_DISTANCE
> - ANDROID_LENS_INFO_MINIMUM_FOCUS_DISTANCE
> - ANDROID_SENSOR_REFERENCE_ILLUMINANT1
> - ANDROID_SENSOR_REFERENCE_ILLUMINANT2
> - ANDROID_SENSOR_CALIBRATION_TRANSFORM1
> - ANDROID_SENSOR_CALIBRATION_TRANSFORM2
> - ANDROID_SENSOR_COLOR_TRANSFORM1
> - ANDROID_SENSOR_COLOR_TRANSFORM2
> - ANDROID_SENSOR_FORWARD_MATRIX1
> - ANDROID_SENSOR_FORWARD_MATRIX2

Does this library use those fields? When will be an update ready with this permission?

natario1, 2019-03-23:
We're not using CameraCharacteristics at all, but if we were, we would not be reading these fields anyway.

But for CameraView to work, you still need the Camera permission!

hardysim, 2019-03-25:
Ah true. I was confused because this permission didn't show up in the merged manifest (but it does when I analyze the final APK).

closed 2019-03-25 by hardysim