natario1/CameraView #422: Use library without support lib

janlukes, 2019-03-25:
### Describe the bug
Documentation states the lib can be used without support libraries, but in fact it requires. 
Build fails on: 
e: Supertypes of the following classes cannot be resolved. Please make sure you have the required dependencies in the classpath:
    class com.otaliastudios.cameraview.CameraView, unresolved supertypes: androidx.lifecycle.LifecycleObserver

- CameraView version: 2.0.0-beta04

### Expected behavior
Use library without support libs. 

I guess you would have to release another lib flavour without support libs.

natario1, 2019-03-25:
We depend on some androidx libraries that include LifecycleObserver, have you removed them explicitly from Gradle? If so, you shouldn't.

If not, maybe you have to migrate your project to androidX, see #369

janlukes, 2019-03-25:
I was kind of expecting to have just this library. Maybe the documentation could use some more explanation. With androidX it works as expected.

HassanUsman, 2019-03-28:
I've also stopped using this library due to their dependency on **androidX**. @natario1 kindly make another lib flavor without  **androidX** support libs.

natario1, 2019-03-29:
The project depends on the androidX lifecycle library which we need internally, so there's not much we can do.

Note that when you use `android.useAndroidX=true` to enable androidX, you are not adding libraries to your project - it is just a compiler flag so everything works correctly.

closed 2019-04-25 by stale[bot]