natario1/CameraView #424: camera preview height not fill parent  with samsung s8 (5.8")

hazemhamadaqa, 2019-03-26:
when i am recording with samsung s8 (5.8") the camera preview height not fill parent  as below picture 

<com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:keepScreenOn="true"
        android:layout_width="match_parent"
        android:layout_height="match_parent" />

### i am used 1.6.0

[image]

GianfrancoMS, 2019-03-26:
I also have the same issue with other Samsung models.

GianfrancoMS, 2019-03-26:
I have the same issue on some Samsung devices.

natario1, 2019-03-29:
It might be a problem with your layout, or maybe your app/activity is not hardware accelerated. Anyway I can't give support for v1. Please migrate to v2, and if you still have this issue, you can open a new issue and fill the bug form.

closed 2019-03-29 by natario1