natario1/CameraView #425: Non-fatal Exception: com.otaliastudios.cameraview.CameraException

huseyinWA, 2019-03-26:
### Describe the bug
My users often get this non-fatal error. It may be my mistake, I couldn't reproduce the error. I just want to inform the community.

Non-fatal Exception: com.otaliastudios.cameraview.CameraException
       at com.otaliastudios.cameraview.Camera1.onVideoResult(Camera1.java:649)
       at com.otaliastudios.cameraview.VideoRecorder.dispatchResult(VideoRecorder.java:28)
       at com.otaliastudios.cameraview.FullVideoRecorder.stop(FullVideoRecorder.java:132)
       at com.otaliastudios.cameraview.Camera1$16.run(Camera1.java:786)
       at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:62)
       at android.os.Handler.handleCallback(Handler.java:789)
       at android.os.Handler.dispatchMessage(Handler.java:98)
       at android.os.Looper.loop(Looper.java:164)
       at android.os.HandlerThread.run(HandlerThread.java:65)

- CameraView version: 2.0.0-beta03
- Reproducible in official demo app: *no*
- Device / Android version: *Many device and OS*

natario1, 2019-03-29:
I don't know what this is about, video might fail for a number of reason, but I will improve the error messages.

HBiSoft, 2019-04-23:
@natario1 I've had 11 users in the past week with this issue and there is no way of knowing what is causing it.

HBiSoft, 2019-04-23:
@huseyinWA check if you can reproduce this by starting the video and stopping it shortly after starting it (within a second of starting it)?

HBiSoft, 2019-04-23:
What I have done temporarily is to make the "stop button" not clickable - `stopRec.setClickable(false);` for 3 seconds inside a `Thread`, as shown below (I added comments to describe what I'm doing):

    startRec.setOnClickListener(new View.OnClickListener() {
            @Override
            public void onClick(View v) {

                // Make the start button invisible / gone
                startRec.setVisibility(View.GONE);

                //Make the stop button visible
                stopRec.setVisibility(View.VISIBLE);

                // I change the resource of my button to this when recording started
                stopRec.setImageResource(R.drawable.record_vid_started);

                // Then stop button cannot be clicked
                stopRec.setClickable(false);

                //Start video recording
                startCaptureVideo();

                //Set animation to stop button
                stopRec.startAnimation(anim);

                //Start thread
                theadWaitBeforeStop = new Thread() {
                    @Override
                    public void run() {
                        try {
                            synchronized (this) {

                                //Wait 3 seconds
                                wait(3000);

                                runOnUiThread(new Runnable() {
                                    @Override
                                    public void run() {
                                        // After 3 seconds I change the resource, indicating that the stop button can be pressed
                                        stopRec.setImageResource(R.drawable.record_vid_btn_stop);

                                        // I make the stop button clickable again
                                        stopRec.setClickable(true);

                                        // Then I stop the thread
                                        theadWaitBeforeStop.interrupt();

                                    }
                                });

                            }
                        } catch (InterruptedException e) {
                            e.printStackTrace();
                        }

                    }
                };
                theadWaitBeforeStop.start();
            }
        });

Doing it this way, it is imposable to stop the recording for 3 seconds, in turn will prevent the issue we had. My device's camera prevents me to press the stop button for 1 second, so 3 seconds might be too much, but I will test the times.

Though this resolves the issue (I hope) we experienced, it would still be nice if this was implemented into the library.

HBiSoft, 2019-05-17:
@natario1 I have updated my app with the above logic, forcing the video not to be stoped within the first 3 seconds of starting the recording. I still get countless logs with this issues.

I even asked one of my users, that contacted me about this issue, to try and record a video of at-least 30 second and check if the error persisted and it did.

I'm at a cross road with this and currently I get more then 10 devices a day that experience this, here is a list of  devices that experienced this the last 2 days:

Samsung Galaxy J7
Samsung Galaxy S5
Samsung Galaxy S8
Samsung Galaxy J5
Samsung Tab A 8
Kameron iTab X Plus 10
OnePlus 6
Google Pixel 2 XL
and 3 other unknown devices

I have improved the error message, as you asked, I tried fixing it by forcing the user not to stop the video for the first 3 second, can you please reply with any suggestions?

HBiSoft, 2019-05-18:
@natario1 
I got more info from the user that's helping me debug this issue, apparently `onCameraError` is called as soon as the user starts recording, I will provide my implementation below to make sure I haven't made a mistake on my end.

I added comments to make reading/understand my logic easier:

    public class camView extends AppCompatActivity {
        CameraView cameraView;
        ImageView startRec, stopRec, flipCam;
        String filename;
        File directoryToStore;
        File filePlusName;
        ProgressBar progress_loader;
        RelativeLayout rel_startstop;
        TextView determRote;
        Thread thread, theadWaitBeforeStop;
        String showDialog;

        @Override
        protected void onCreate(Bundle savedInstanceState) {
            super.onCreate(savedInstanceState);
            getWindow().setFlags(WindowManager.LayoutParams.FLAG_HARDWARE_ACCELERATED,
                WindowManager.LayoutParams.FLAG_HARDWARE_ACCELERATED);
            setContentView(R.layout.activity_cam_view);
            //not importatnt
            showDialog = getIntent().getStringExtra("showDialog");

            initViews();

            cameraView.setLifecycleOwner(this);
            cameraView.setMode(Mode.VIDEO);

            //SET LOG LEVEL
            CameraLogger.setLogLevel(CameraLogger.LEVEL_VERBOSE);
            CameraLogger.registerLogger(new CameraLogger.Logger() {
                @Override
                public void log(int level, @NonNull String tag, @NonNull String message, @Nullable Throwable throwable) {
                    Crashlytics.log(level, tag, message);

                }
            });

            //First make buttons invisible
            rel_startstop.setVisibility(View.INVISIBLE);
            flipCam.setVisibility(View.INVISIBLE);
            //then show them
            final Animation anim = new AlphaAnimation(0.0f, 1.0f);
            animationToShowButtons(anim);

            //FILE NAME AND PATH
            @SuppressLint("SimpleDateFormat") SimpleDateFormat formatter = new SimpleDateFormat("yyyy_MM_dd_HH_mm_ss");
            Date now = new Date();
            filename = formatter.format(now) + ".mp4";
            directoryToStore = getBaseContext().getExternalFilesDir("CameraTemp");
            filePlusName = new File(directoryToStore, filename);

            //CHANGE CAMERA - FRONT/BACK
            flipCam.setOnClickListener(new View.OnClickListener() {
                @Override
                public void onClick(View v) {
                    if (getApplicationContext().getPackageManager().hasSystemFeature(PackageManager.FEATURE_CAMERA_FRONT)) {
                        if (cameraView.getFacing() == Facing.FRONT) {
                            cameraView.setFacing(Facing.BACK);
                        } else {
                            cameraView.setFacing(Facing.FRONT);
                        }
                    } else {
                        Toast.makeText(camView.this, "Your device doesn't have a front camera..", Toast.LENGTH_SHORT).show();
                    }

                }
[... 193 lines cut ...]

        //START VIDEO RECORDING
        private void startCaptureVideo() {
            try {
                if (!cameraView.isTakingVideo()) {
                    cameraView.takeVideo(filePlusName);
                }
            } catch (Exception e) {
                Crashlytics.logException(e);
            }
        }

        //STOP VIDEO RECORDING
        private void stopCaptureVideo() {
            cameraView.stopVideo();
            progress_loader.setVisibility(View.VISIBLE);
        }

        @Override
        protected void onDestroy() {
            super.onDestroy();
            cameraView.destroy();
        }

    }

commit 1fef9d28 (2019-05-25): Fix #425
commit 8b66d5b5 (2019-05-25): Quick fixes & bump version (#471)

* Fix #460

* Fix #419

* Fix #443

* Bump version

* Fix #425

* Fix test
closed 2019-05-25 by natario1