natario1/CameraView #430: Frame Processor stops running

jgranleese953, 2019-03-31:
### Describe the bug
I've implemented a custom Frame Processor that I attach to the CameraView and "onProcess" is called successfully for the first few frames but then the callback returns a frame with data that is null (same error as [here](https://github.com/natario1/CameraView/issues/423#issuecomment-478377593)) and therefore is unable to freeze. After that the callback stops returning at all.

- CameraView version: 1.6.1
- Device / Android version: Samsung S9, API 28

jgranleese953, 2019-04-01:
### **FIX**
`@Override
        public void dispatchFrame(final Frame frame) {
            if (mFrameProcessors.isEmpty()) {
                // Mark as released. This instance will be reused.
                frame.release();
            } else {
                mLogger.v("dispatchFrame:", frame.getTime(), "processors:", mFrameProcessors.size());
                mFrameProcessorsHandler.post(new Runnable() {
                    @Override
                    public void run() {
                        for (FrameProcessor processor : mFrameProcessors) {
                            processor.process(frame);
                        }
                        frame.release();
                    }
                });
            }
        }`

Should be:

`@Override
        public void dispatchFrame(final Frame frame) {
            if (mFrameProcessors.isEmpty()) {
                // Mark as released. This instance will be reused.
                frame.release();
            } else {
                mLogger.v("dispatchFrame:", frame.getTime(), "processors:", mFrameProcessors.size());
                mFrameProcessorsHandler.post(new Runnable() {
                    @Override
                    public void run() {
                        for (FrameProcessor processor : mFrameProcessors) {
                            processor.process(frame);
                        }
                    }
                });
            }
        }`

The "  frame.release();" method call causes a race condition

closed 2019-04-02 by jgranleese953

hanspeide, 2019-04-03:
@jgranleese953 Why was this issue closed?

jgranleese953, 2019-04-04:
Remove [this line](https://github.com/natario1/CameraView/blob/fc4a75a24b43d0b72a265cfc060f4e052d2ccea5/cameraview/src/main/java/com/otaliastudios/cameraview/CameraView.java#L1811) to resolve the issue