natario1/CameraView #440: onCameraError called after onVideoTaken + com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Error inside the onError callback.

mittt, 2019-04-12:
### Describe the bug
onCameraError is called after onVideoTaken, with the following exception:

```
com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Error inside the onError callback. 1
        at com.otaliastudios.cameraview.Camera1.onError(Camera1.java:348)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1146)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:148)
        at android.os.HandlerThread.run(HandlerThread.java:61)
     Caused by: java.lang.RuntimeException: Error inside the onError callback. 1
        at com.otaliastudios.cameraview.Camera1.onError(Camera1.java:341)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1146) 
        at android.os.Handler.dispatchMessage(Handler.java:102) 
        at android.os.Looper.loop(Looper.java:148) 
        at android.os.HandlerThread.run(HandlerThread.java:61) 
```

This is happening in our own app and the demo app, also. I'll attach the LogCat from the demo app.

- CameraView version: com.otaliastudios:cameraview:2.0.0-beta04
- Reproducible in official demo app: yes - with r783
- Device / Android version: Asus Google Nexus 7 (2013), Android 6.0.1, Build MOB30X

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Run the demo app
2. Change the mode to video
3. Record video
4. After the 5 sec, there is a 3-4 seconds freeze, the preview is played and a "Got CameraException #0" toast

### Expected behavior
onCameraError  not called after successful recording

### Logs
https://gist.github.com/mittt/634839088b95243436885e82659d7826

### APK
Link to a Github repo where the bug is reproducible:
https://github.com/natario1/CameraView.git/trunk

woutergoossens, 2019-05-05:
Have the same issue. How can this be fixed?

I have a pause functionality in my video recording screen. So I stop the video recording but then also the preview stops and I get this error. 

When I want to open the camera again, it doesn’t work (because of the error)

woutergoossens, 2019-05-05:
05-05 21:51:43.449 2683-2699/com.otaliastudios.cameraview.demo E/Camera: Error 1
05-05 21:51:43.449 2683-2699/com.otaliastudios.cameraview.demo E/Camera1: Error inside the onError callback. 1
05-05 21:51:43.452 2683-2699/com.otaliastudios.cameraview.demo E/CameraController: uncaughtException: Interrupting thread with state: STATE_STARTED due to CameraException: com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Error inside the onError callback. 1
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Error inside the onError callback. 1
        at com.otaliastudios.cameraview.Camera1.onError(Camera1.java:347)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1146)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:148)
        at android.os.HandlerThread.run(HandlerThread.java:61)
     Caused by: java.lang.RuntimeException: Error inside the onError callback. 1

natario1, 2019-05-23:
Will improve error messages in #425 ... Without this there's not much we can do. If you debug yourself and find an issue, please reopen or open a new issue

closed 2019-05-23 by natario1