natario1/CameraView #442: What should I do if I want to rotate the lens?

oksimple, 2019-04-13:
What should I do if I want to rotate the lens?

oksimple, 2019-04-14:
Rotate the picture instead of adjusting the lens.

oksimple, 2019-04-14:
I tried to modify CameraController. offset (), but the modification would affect addFrameProcessor. Do you have any good suggestions?

wosiu, 2019-04-20:
You don't need to modify CameraController to get rotated image. Just postprocess a picture / snapshot / frame, once you get it. So first convert bytes to bitmap and then use proper Matrix transformation, e.g.:
```
Matrix matrix = new Matrix();
matrix.postRotate(90);
Bitmap rotatedBitmap = Bitmap.createBitmap(orgBitmap, 0, 0, scaledBitmap.getWidth(), scaledBitmap.getHeight(), matrix, true);
```
Also there is a ready util function somewhere in this lib for that by Natario. There are also few more options for rotating (-> google), but if you want to just rotate - mentioned ones are fine.

oksimple, 2019-04-22:
@wosiu I'm glad to hear from you.  You're talking about spinning after taking a picture, I know, but my need is to preview and adjust the lens.

closed 2019-05-19 by stale[bot]