natario1/CameraView #446: Orientation/rotation always 0

wosiu, 2019-04-22:
### Describe the bug
When I take picture or snapshot while device is rotated, photo I get in result is already rotated to match device rotation. However `rotation` is set to 0 all the time. Thus there is no way currently to find out if I have to unrotate an image if I want to keep absolute position of the image.

- CameraView version: *version number*
1.6.1 and 2.0.0-beta04 tested
- Reproducible in official demo app: *yes/no*
yes
- Devices / Android version: *Pixel, API 28*
1. Android 7.0 - Huawei p10 lite
2. Android 7.0 - NOMU s30 mini
3. Android 7.1.2 - Samsung Galaxy S3 (GT-I9300)

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Open demo or your app using the lib
2. Rotate device `X` degrees (where `X`>90)
3. take picture or snapshot
4. Picture you get is already rotated `-X` degrees, and result `rotation` value is set to 0 (both from exif data in 1.6.1 version and `result.getRotation()` in 2.0)

### Expected behavior
If picture is automatically rotated  with respect to the device rotation, `rotation` should be set to proper value, so that I could undo the rotation.
In my case it is crucial to have an absolute picture/snapshot without taking into account the rotation of the device. 

### Screenshots
This is how I capture photo:
[image]
This is what I get:
[image]
I expect either: the photo to be rotated OR `rotation` set to 90

natario1, 2019-04-22:
The meaning of result.getRotation() is that of the EXIF rotation, so if EXIF = 0, result.rotation = 0.

About the jpeg image being automatically rotated (instead of left as is and provided with non-0 EXIF rotation), this happens on some devices and is done by the Android framework, so we basically have no control over it.

Maybe you can use the orientation in CameraCallback.onOrientationChanged

wosiu, 2019-04-22:
Sounds good. Using such rotation I figured out that I need to rotate image with `360-orientation`.
Do you know if I should also add/minus exif orientation from `result` (or `Utils.getExifRotation(jpeg)`) to the final formula? It sounds for me I could just skip it.

wosiu, 2019-05-08:
Answering my own question for others - I ended up with formula:

`int compensation = (360 - deviceRotation + exifRotation) % 360;`

where `deviceRotation` is from `onOrientationChanged` callback 
and `int exifRotation = Utils.getExifRotation(jpeg)` (or `result.getRotation()` in v2.0).

That part with `exifRotation` was required e.g. for Sony Xperia M5.

closed 2019-05-08 by wosiu

mariuspena, 2019-05-13:
@wosiu hey there. 
when you say deviceRotation you are referring to the one you get like this:
int rotation = getWindowManager().getDefaultDisplay().getRotation(); // values: 0, 1, 2, 3
or to the angle (possible values 0..360) you receive on onOrientationChanged ?

wosiu, 2019-05-14:
@mariuspena
to the angle (possible values 0..360) received on onOrientationChanged

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Wolny
od wirusów. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>