natario1/CameraView #448: LifecycleObserver not found

Divay-Parashar, 2019-04-23:
### How do I?
when I write this line 
 camera.setLifecycleOwner(this);
in onCreate method and compile my code this says
error: cannot access LifecycleObserver
the class file for androidx.lifecycle.LifecycleObserver not found. How do I fix this error?

### Version used
CameraView exact version.
implementation 'com.otaliastudios:cameraview:2.0.0-beta04'

yelsane, 2019-05-01:
You're probably not using the AndroidX Support Libraries. There's a note in the [documentation](https://natario1.github.io/CameraView/about/getting-started.html):

> Can’t resolve the lifecycle owner interface? Read [below](https://natario1.github.io/CameraView/about/getting-started.html#without-support-libraries).

Either add the support library or implement it as shown in the docs without the libraries.

natario1, 2019-05-23:
Yes, you should migrate your app to AndroidX if you want to use CameraView v2.

closed 2019-05-23 by natario1