natario1/CameraView #451: CameraView demo error on start and takePicture()

kleysonr, 2019-04-25:
Device: AVD Nexus 6 API 26

Please see **ERROR 1** and **ERROR 2** below.

**ERROR 1**

When the CameraView demo app is initialized I get in the Logcat the following error. Also, in the app I see a black background instead of the camera image.

But I if switch the camera twice the app starts to show the camera image.

```
2019-04-24 21:21:27.420 9954-9954/? I/zygote: Not late-enabling -Xcheck:jni (already on)
2019-04-24 21:21:27.429 9954-9954/com.smartnsens.cameraviewtest W/zygote: Unexpected CPU variant for X86 using defaults: x86
2019-04-24 21:21:27.540 9954-9954/com.smartnsens.cameraviewtest I/InstantRun: starting instant run server: is main process
2019-04-24 21:21:27.676 9954-9954/com.smartnsens.cameraviewtest I/zygote: Rejecting re-init on previously-failed class java.lang.Class<androidx.core.view.ViewCompat$2>: java.lang.NoClassDefFoundError: Failed resolution of: Landroid/view/View$OnUnhandledKeyEventListener;
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at void androidx.core.view.ViewCompat.setBackground(android.view.View, android.graphics.drawable.Drawable) (ViewCompat.java:2561)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at void androidx.appcompat.widget.ActionBarContainer.<init>(android.content.Context, android.util.AttributeSet) (ActionBarContainer.java:62)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at java.lang.Object java.lang.reflect.Constructor.newInstance0(java.lang.Object[]) (Constructor.java:-2)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at java.lang.Object java.lang.reflect.Constructor.newInstance(java.lang.Object[]) (Constructor.java:334)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at android.view.View android.view.LayoutInflater.createView(java.lang.String, java.lang.String, android.util.AttributeSet) (LayoutInflater.java:647)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at android.view.View android.view.LayoutInflater.createViewFromTag(android.view.View, java.lang.String, android.content.Context, android.util.AttributeSet, boolean) (LayoutInflater.java:790)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at android.view.View android.view.LayoutInflater.createViewFromTag(android.view.View, java.lang.String, android.content.Context, android.util.AttributeSet) (LayoutInflater.java:730)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at void android.view.LayoutInflater.rInflate(org.xmlpull.v1.XmlPullParser, android.view.View, android.content.Context, android.util.AttributeSet, boolean) (LayoutInflater.java:863)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at void android.view.LayoutInflater.rInflateChildren(org.xmlpull.v1.XmlPullParser, android.view.View, android.util.AttributeSet, boolean) (LayoutInflater.java:824)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at android.view.View android.view.LayoutInflater.inflate(org.xmlpull.v1.XmlPullParser, android.view.ViewGroup, boolean) (LayoutInflater.java:515)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at android.view.View android.view.LayoutInflater.inflate(int, android.view.ViewGroup, boolean) (LayoutInflater.java:423)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at android.view.View android.view.LayoutInflater.inflate(int, android.view.ViewGroup) (LayoutInflater.java:374)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at android.view.ViewGroup androidx.appcompat.app.AppCompatDelegateImpl.createSubDecor() (AppCompatDelegateImpl.java:608)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at void androidx.appcompat.app.AppCompatDelegateImpl.ensureSubDecor() (AppCompatDelegateImpl.java:519)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at void androidx.appcompat.app.AppCompatDelegateImpl.setContentView(int) (AppCompatDelegateImpl.java:467)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at void androidx.appcompat.app.AppCompatActivity.setContentView(int) (AppCompatActivity.java:141)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at void com.smartnsens.cameraviewtest.CameraActivity.onCreate(android.os.Bundle) (CameraActivity.java:38)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at void android.app.Activity.performCreate(android.os.Bundle) (Activity.java:6975)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at void android.app.Instrumentation.callActivityOnCreate(android.app.Activity, android.os.Bundle) (Instrumentation.java:1213)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at android.app.Activity android.app.ActivityThread.performLaunchActivity(android.app.ActivityThread$ActivityClientRecord, android.content.Intent) (ActivityThread.java:2770)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at void android.app.ActivityThread.handleLaunchActivity(android.app.ActivityThread$ActivityClientRecord, android.content.Intent, java.lang.String) (ActivityThread.java:2892)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at void android.app.ActivityThread.-wrap11(android.app.ActivityThread, android.app.ActivityThread$ActivityClientRecord, android.content.Intent, java.lang.String) (ActivityThread.java:-1)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at void android.app.ActivityThread$H.handleMessage(android.os.Message) (ActivityThread.java:1593)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at void android.os.Handler.dispatchMessage(android.os.Message) (Handler.java:105)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at void android.os.Looper.loop() (Looper.java:164)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at void android.app.ActivityThread.main(java.lang.String[]) (ActivityThread.java:6541)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at java.lang.Object java.lang.reflect.Method.invoke(java.lang.Object, java.lang.Object[]) (Method.java:-2)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at void com.android.internal.os.Zygote$MethodAndArgsCaller.run() (Zygote.java:240)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at void com.android.internal.os.ZygoteInit.main(java.lang.String[]) (ZygoteInit.java:767)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote: Caused by: java.lang.ClassNotFoundException: Didn't find class "android.view.View$OnUnhandledKeyEventListener" on path: DexPathList[[zip file "/data/app/com.smartnsens.cameraviewtest-hvfJ3Y8ts4EEi24PGpByVQ==/base.apk", zip file "/data/app/com.smartnsens.cameraviewtest-hvfJ3Y8ts4EEi24PGpByVQ==/split_lib_dependencies_apk.apk", zip file "/data/app/com.smartnsens.cameraviewtest-hvfJ3Y8ts4EEi24PGpByVQ==/split_lib_resources_apk.apk", zip file "/data/app/com.smartnsens.cameraviewtest-hvfJ3Y8ts4EEi24PGpByVQ==/split_lib_slice_0_apk.apk", zip file "/data/app/com.smartnsens.cameraviewtest-hvfJ3Y8ts4EEi24PGpByVQ==/split_lib_slice_1_apk.apk", zip file "/data/app/com.smartnsens.cameraviewtest-hvfJ3Y8ts4EEi24PGpByVQ==/split_lib_slice_2_apk.apk", zip file "/data/app/com.smartnsens.cameraviewtest-hvfJ3Y8ts4EEi24PGpByVQ==/split_lib_slice_3_apk.apk", zip file "/data/app/com.smartnsens.cameraviewtest-hvfJ3Y8ts4EEi24PGpByVQ==/split_lib_slice_4_apk.apk", zip file "/data/app/com.smartnsens.cameraviewtest-hvfJ3Y8ts4EEi24PGpByVQ==/s
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at java.lang.Class dalvik.system.BaseDexClassLoader.findClass(java.lang.String) (BaseDexClassLoader.java:93)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String, boolean) (ClassLoader.java:379)
2019-04-24 21:21:27.677 9954-9954/com.smartnsens.cameraviewtest I/zygote:     at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String) (ClassLoader.java:312)
[... cut ...]
        at android.app.ActivityThread.-wrap1(Unknown Source:0)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1661)
        at android.os.Handler.dispatchMessage(Handler.java:105)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6541)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)
```

kleysonr, 2019-04-25:
Might be related with the issue #450

kleysonr, 2019-04-25:
Regarding the **ERROR 2** I've found this issue #191 

Please include in the documentation or in the build.gradle (app) of the CameraView demo the configuration:

```implementation 'androidx.exifinterface:exifinterface:1.0.0'```

Best Regards.

kleysonr, 2019-05-15:
Issues being closed automatically without any answer from the developers.

natario1, 2019-05-23:
You can see that androidx.exifinterface is already present in cameraview's build.gradle file. I'm not sure  why you get this error, might be an instant run issue or something similar... try to clean your project and restart.

kleysonr, 2019-06-13:
Hi @natario1 

The 'androidx.exifinterface' is not present in the 'demo'.

Best Regards.
Kleyson Rios.

kleysonr, 2019-07-03:
To keep the issue opened.

closed 2019-07-30 by stale[bot]