natario1/CameraView #452: Is it possible disable the CameraView preview in order to only use FrameProcessor ?

kleysonr, 2019-04-25:
CameraView version: v2.0.0-beta04

I would like to only use the FrameProcessor and render each frame directly in a ImageView.

And to avoid a waste of resource in the device, is it possible, somehow, disable the rendering of the preview frames in the <com.otaliastudios.cameraview.CameraView> and only work with the frames inside the FrameProcessor ?

Best Regards.

kleysonr, 2019-05-15:
Hi, @natario1 

Any help on this ?

Best Regards.

natario1, 2019-05-23:
I don't think so @kleysonr ! You can try. But rendering each frame as a Bitmap in a ImageView is not very efficient...

closed 2019-05-23 by natario1

kleysonr, 2019-06-13:
Hi @natario1 

Thanks for the reply, but what do you suggest as a good approach to render each frame again after it be processed by the FrameProcessor ?

Best Regards.