natario1/CameraView #455: Orientation automatically changes after i take image on some devices

sandeep7644, 2019-04-27:
When i am taking images from front camera orientation automatically changes to landscape mode.
Please suggest me something to fix this issue.

wosiu, 2019-05-08:
Maybe:
```
int rotationCompensation = Utils.getExifRotation(jpegBytes);
Matrix trans = new Matrix();
trans.setRotate(rotationCompensation);
```
and rotate bitmap decoded from `jpegBytes` using `trans` matrix

wosiu, 2019-05-08:
Not sure if you also might need a device rotation for this. I had similar issue with back camera -> #446

sandeep7644, 2019-05-09:
> Maybe:
> 
> ```
> int rotationCompensation = Utils.getExifRotation(jpegBytes);
> Matrix trans = new Matrix();
> trans.setRotate(rotationCompensation);
> ```
> 
> and rotate bitmap decoded from `jpegBytes` using `trans` matrix

where is Utils ??

sandeep7644, 2019-05-15:
I used takeSnapShot instead of takePicture and the issue is resolved.

closed 2019-05-15 by sandeep7644