natario1/CameraView #457: how to install this package.

johnjwatson, 2019-04-27:
I am quite new to android dev (C engineer here), and I am not able to get this to install.
 My dependency block:

    dependencies {
        implementation fileTree(dir: 'libs', include: ['*.jar'])
        implementation"org.jetbrains.kotlin:kotlin-stdlib-jdk7:$kotlin_version"
        implementation 'com.android.support:appcompat-v7:28.0.0'
implementation 'com.android.support.constraint:constraint-layout:1.1.3'
implementation 'com.otaliastudios:cameraview:2.0.0-beta04'
testImplementation 'junit:junit:4.12'
androidTestImplementation 'com.android.support.test:runner:1.0.2'
androidTestImplementation 'com.android.support.test.espresso:espresso-core:3.0.2'
    }

Main gradle file:

    allprojects {
    repositories {
        google()
        jcenter()
        
        }
    }

Then I sync with gradle files and I still get unreslved reference!!!

I looked at the doc:

https://natario1.github.io/CameraView/about/install.html

    Then simply download the latest version:
    api 'com.otaliastudios:cameraview:2.0.0-beta04'

I tried putting this in the dependencies but still same result.

wosiu, 2019-05-08:
Could you please paste build logs with unresolved references.
However I'm afraid in advance this is not an issue related with CameraView lib.
I suggested starting with demo project if you're new to gradle: https://github.com/natario1/CameraView/blob/master/demo/build.gradle
also migration from `com.android.support` to `androidx` is recommended.

natario1, 2019-05-23:
Like @wosiu said!

closed 2019-05-23 by natario1