natario1/CameraView #458: Pinch to Zoom not woking smoothly in Android 9.0

Pinkal7600, 2019-04-29:
I have checked a camera view in my app and also with the demo app. While I perform pinch to zoom, camera preview is stacking and take some time to stable preview. I have tested in different devices with Android OS 6.0, 7.0, 8.0 and 9.0. I only find this issue in Android 9.0 and in other Android versions it works smoothly.

Video link: https://drive.google.com/file/d/1o61peUf3xreqEhrvbhh0wLBAW2LVQfBd/view?usp=sharing

Pinkal7600, 2019-05-20:
Anyone have check this issue and face in demo app?

natario1, 2019-05-23:
I'm not sure how to help @Pinkal7600 ! If you can spot an issue with code let me know. If not follow #217 , we might add some control to change the speed of this

closed 2019-05-23 by natario1