natario1/CameraView #462: Runtime Exception Not Handled for Draw

looplearn, 2019-05-09:
Hi, I wanted to log the following issue in relation to eGL rendering for very long time periods.
I am unsure of condition to reproduce the error, but the following error is happening on several devices.  It appears to happen after a wake lock transition, but cannot be sure.  The application runs close to 24/7 on several devices and we have under ten of these unhandled exceptions over the last couple of weeks.
Fatal Exception: java.lang.RuntimeException
Error during glDrawArrays glError 0x 505
com.otaliastudios.cameraview.EglElement.check (EglElement.java:26)
com.otaliastudios.cameraview.EglViewport.drawFrame (EglViewport.java:176)
com.otaliastudios.cameraview.EglViewport.drawFrame (EglViewport.java:120)
com.otaliastudios.cameraview.GlCameraPreview.onDrawFrame (GlCameraPreview.java:207)
android.opengl.GLSurfaceView$GLThread.guardedRun (GLSurfaceView.java:1571)
android.opengl.GLSurfaceView$GLThread.run (GLSurfaceView.java:1270)

looplearn, 2019-05-09:
#Describe the bug
Unhandled Exception Thrown when drawing on eGL surface when run for long periods of time.

CameraView version: 2.0.0-beta04
Reproducible in official demo app: No
Device / Android version: Rockchip Nano PC-T4, API 27
#To Reproduce
Remove Application from battery optimisation list (not optimised for battery / always on)
Run application for very long time (happens more often if power supply or battery is low)
Attempt to draw bitmap of image from frame processor on same surface as preview.

#Expected Behaviour
 Expected "onCameraError" Event instead of throwing unhandled exception

natario1, 2019-05-23:
0x505 seems to be a GL_OUT_OF_MEMORY error... CameraView will reuse the same memory over and over so I am not sure there's nothing we can do. Make sure you are not allocating new memory at each frame.

closed 2019-06-20 by stale[bot]