natario1/CameraView #466: Preview stretches for a moment while switching front/back

dhruv2295, 2019-05-15:
### Describe the bug
A clear and concise description of what the bug is.

- CameraView version: *2.0.0-beta04*
- Reproducible in official demo app: *yes*
- Device / Android version: *Pixel 3, API 28*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Select Video Mode
2. Set width and height to match_parent
3. Click on toggle camera button

### Expected behavior
While switching the preview screen stretches for a second

natario1, 2019-05-24:
I'm not sure this can be fixed... But will take a look.

MoustafaElsaghier, 2019-08-09:
Mr @natario1 any updates on this issue?

jlertele, 2019-12-19:
I have the same problem...

vaidik1011, 2021-09-28:
It will probably not stretches if you give camera view ratio of 4:3. maybe it's not the solution but you can overcome that problem using it.

SweetD3v, 2022-04-26:
Yes, It won't stretch if you keep the same aspect ratio on either of the cameras.

xiaoyaomeng, 2022-07-05:
I have the same problem...