natario1/CameraView #467: setParameters failed and CameraException in onVideoResult

HBiSoft, 2019-05-19:
I actually have 2 error, they might be connected, @natario1 might have seen me post on multiple issues.

According to a user that contacted me, `onError` is called as soon as he starts recording a video.

- CameraView version: *2.0.0-beta03*
- Reproducible in official demo app: *app is in production*

Devices / Android version: 

- Samsung Galaxy J7
- Samsung Galaxy S5
- Samsung Galaxy S8
- Samsung Galaxy J5
- Samsung Tab A 8
- Kameron iTab X Plus 10
- OnePlus 6
- Google Pixel 2 XL
- and 3 other unknown devices

**Here is the 2 logs I get in Crashlytics:**

---
**First**

Stacktrack:

    Fatal Exception: java.lang.RuntimeException: setParameters failed
       at android.hardware.Camera.native_setParameters(Camera.java)
       at android.hardware.Camera.setParameters(Camera.java:2171)
       at com.otaliastudios.cameraview.Camera1.createCamera(Camera1.java:242)
       at com.otaliastudios.cameraview.Camera1.onStart(Camera1.java:266)
       at com.otaliastudios.cameraview.CameraController$3.run(CameraController.java:183)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:135)
       at android.os.HandlerThread.run(HandlerThread.java:61)

Log:

    CameraController Start: posting runnable. State: STATE_STOPPED
    CameraController Start: executing. State: STATE_STOPPED
    CameraController Start: about to call onStart() STATE_STARTING
    Camera1 collectCameraId Facing: BACK Internal: 0 Cameras: 2
    CameraCallbacks onDeviceOrientationChanged 0
    CameraView preview: isHardwareAccelerated: true
    CameraView onMeasure: surface is not ready. Calling default behavior.
    CameraView onMeasure: surface is not ready. Calling default behavior.
    CameraView onMeasure: surface is not ready. Calling default behavior.
    CameraView onMeasure: surface is not ready. Calling default behavior.
    CameraPreview dispatchOnOutputSurfaceAvailable: w= 1200 h= 1824

I know the above log doesn't give much info, but that is what I get in Crashlytics.

---

**Second**

Stacktrack:

    Non-fatal Exception: com.otaliastudios.cameraview.CameraException
       at com.otaliastudios.cameraview.Camera1.onVideoResult(Camera1.java:649)
       at com.otaliastudios.cameraview.VideoRecorder.dispatchResult(VideoRecorder.java:28)
       at com.otaliastudios.cameraview.FullVideoRecorder.stop(FullVideoRecorder.java:132)
       at com.otaliastudios.cameraview.FullVideoRecorder.start(FullVideoRecorder.java:108)
       at com.otaliastudios.cameraview.Camera1$14.run(Camera1.java:691)
       at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:62)
       at android.os.Handler.handleCallback(Handler.java:790)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at android.os.Looper.loop(Looper.java:164)
       at android.os.HandlerThread.run(HandlerThread.java:65)
[... 313 lines cut ...]
                }
            } catch (Exception e) {
                Crashlytics.logException(e);
            }
        }

        //STOP VIDEO RECORDING
        private void stopCaptureVideo() {
            cameraView.stopVideo();
            progress_loader.setVisibility(View.VISIBLE);
        }

        @Override
        protected void onDestroy() {
            super.onDestroy();
            cameraView.destroy();
        }

    }

It's not related to permissions, I ask permissions and provide permissions correctly.

---

I have also added a lot of info about what I've tried thus far in #425 (At the bottom).

HBiSoft, 2019-05-20:
@natario1 
I have more info, sort of..

I setup a demo application and sent it to the user that is experiencing the issue, I logged the following in `onError`:

    int errorReason = exception.getReason();
    String exceptionAsString = exception.toString();
    String exceptionAsMessage = exception.getMessage();

and it returned:

    5 - meaning REASON_VIDEO_FAILED
    com.otaliastudios.cameraview.CameraException
    null

I'm not sure what else I can do now... My users can't record videos and I can't find out why.

natario1, 2019-05-23:
@HBiSoft as said in #425 we should improve error messages here to know what's going on... I think you should try to inspect the exception thrown in FullVideoRecorder line 111, where the recorder itself catches a start exception and calls stop(). 

That exception is not logged anywhere so we don't know what's the issue that MediaRecorder is complaining about. Most likely some parameters like maxDuration...

HBiSoft, 2019-05-24:
@natario1  Thank you for your reply.

After looking at FullVideoRecorder, I noticed the following in `start`:

    mMediaRecorder.setOnInfoListener(new MediaRecorder.OnInfoListener() {
        @Override
        public void onInfo(MediaRecorder mediaRecorder, int what, int extra) {
                switch (what) {
                case MediaRecorder.MEDIA_RECORDER_INFO_MAX_DURATION_REACHED:
                    mResult.endReason = VideoResult.REASON_MAX_DURATION_REACHED;
                    stop();
                    break;
                case MediaRecorder.MEDIA_RECORDER_INFO_MAX_FILESIZE_REACHED:
                    mResult.endReason = VideoResult.REASON_MAX_SIZE_REACHED;
                    stop();
                    break;
            }
       }
    });

So you are right, since `onError` gets called as soon as the record button is clicked, it could be one of the following reasons, either `MediaRecorder.MEDIA_RECORDER_INFO_MAX_DURATION_REACHED` or `MediaRecorder.MEDIA_RECORDER_INFO_MAX_FILESIZE_REACHED`.

But the thing is, I don't set max duration or max file size, so there shouldn't be a max file size or max duration?

Is there a default set for max duration and max file size?

---

But it's not to say it's definitely the issue because the following is also called in `start()`:

    try {
        mMediaRecorder.prepare();
        mMediaRecorder.start();
    } catch (Exception e) {
        mResult = null;
        stop();
    }

HBiSoft, 2019-05-25:
@natario1 

I see you bumped the version and included the following:

        try {
            mMediaRecorder.prepare();
            mMediaRecorder.start();
        } catch (Exception e) {
            LOG.w("stop:", "Error while starting media recorder.", e);
            mResult = null;
            mError = e;
            stop();
        }

Can you please tell me how I can properly log this issue? Like I mentioned, I have a user that is willing to test and send me the logs. This is how I currently log it:

    public void onCameraError(@NonNull CameraException exception) {
            super.onCameraError(exception);

            int errorReason = exception.getReason();
            String exceptionAsString = exception.toString();
            String exceptionAsMessage = exception.getMessage();

            camView.this.errorList.add("------ exception.getReason() started -----");
            camView.this.errorList.add(String.valueOf(errorReason));
            camView.this.errorList.add("------ exception.getReason() ended ------");
            camView.this.errorList.add("------ exception.toString() started ------");
            camView.this.errorList.add(exceptionAsString);
            camView.this.errorList.add("------ exception.toString() ended ------");
            camView.this.errorList.add("------ exception.getMessage() started ------");
            camView.this.errorList.add(exceptionAsMessage);
            camView.this.errorList.add("------ exception.getMessage() ended ------");
           
            //Then I create a text file that the user can send to me
            camview.generateNoteOnSD(camview.getApplicationContext(), "errorLog.txt");
            
        }

Like I mentioned above, this will return:

    5 - meaning REASON_VIDEO_FAILED
    com.otaliastudios.cameraview.CameraException
    null

which is not very helpful.

So, can you please suggest a way I can log the true issue while I still have a willing user that will sent me the log?

natario1, 2019-05-25:
Yes! You are interested in the source exception, exception.getCause(). Can log toString() and getMessage() of that one.

HBiSoft, 2019-05-26:
@natario1 I added the following in `OnError`:

    @Override
            public void onCameraError(@NonNull CameraException exception) {
                super.onCameraError(exception);
                oldCamView.this.progress_loader.setVisibility(View.GONE);

                int errorReason = exception.getReason();
                String exceptionAsString = exception.toString();
                String exceptionAsMessage = exception.getMessage();
                String exceptionAsCauseAsString = exception.getCause().toString();
                String exceptionAsCauseAsMessage = exception.getCause().getMessage();
                
                errorList.add("exception.getReason() started -----");
                errorList.add(String.valueOf(errorReason));
                errorList.add("exception.getReason() ended ------");

                errorList.add("exception.toString() started ------");
                errorList.add(exceptionAsString);
                errorList.add("exception.toString() ended ------");

                errorList.add("exception.getMessage() started ------");
                errorList.add(exceptionAsMessage);
                errorList.add("exception.getMessage() ended ------");

                errorList.add("exception.getCause().toString() started ------");
                errorList.add(exceptionAsCauseAsString);
                errorList.add("exception.getCause().toString() ended ------");

                errorList.add("exception.getCause().getMessage() started ------");
                errorList.add(exceptionAsCauseAsMessage);
                errorList.add("exception.getCause().getMessage() ended ------");

                generateNoteOnSD( "errorLog.txt");

                showErrorDialog();

            }

Basically just adding all in a Array then I save it to a text file and here is the result:

    exception.getReason() started -----
    5
    exception.getReason() ended ------

    exception.toString() started ------
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: start failed.
    exception.toString() ended ------

    exception.getMessage() started ------
    java.lang.RuntimeException: start failed.
    exception.getMessage() ended ------

    exception.getCause().toString() started ------
    java.lang.RuntimeException: start failed.
    exception.getCause().toString() ended ------

    exception.getCause().getMessage() started ------
    start failed.
    exception.getCause().getMessage() ended ------

Not sure why start fails only on specific devices?

natario1, 2019-05-26:
I don't know :-) MediaRecorder is not liking one of the parameters. It's impossible to tell which one without owning one of these devices and do some trial and error.

Some of these parameters come from CameraView but we do our best to ensure they are supported. Some others come from you (audio bitrate, video bitrate, maxduration, ...) and we can't do any validation.

HBiSoft, 2019-05-26:
@natario1 Thank you for the quick reply. I'm not setting any parameters, not sure what I can do now. As mentioned, it's quite a lot of devices that experience this.

natario1, 2019-05-26:
I don't know! I don't own any. One should have it and launch the demo app and try changing stuff in FullVideoRecorder to spot which parameter is crashing and see how we can improve this

HBiSoft, 2019-05-27:
@natario1 

I implemented OpenCamera to test and I'm not getting any crashes.

You mentioned that  some parameter are causing the crash. Since I'm not getting a crash using OpenCamera I looked at the difference in parameters passed to `MediaRecorder` and the only difference I could see is, in OpenCamera MediaRecorder's AudioSource is set to:

    MediaRecorder.AudioSource.CAMCORDER

and in FullVideoRecorder it is set to:

    MediaRecorder.AudioSource.DEFAULT

Also, in OpenCamera, video size is set like this:

    media_recorder.setVideoSize(camcorderProfile.videoFrameWidth, camcorderProfile.videoFrameHeight)

and in FullVideoRecorder:

    Size size = mResult.getRotation() % 180 != 0 ? mResult.getSize().flip() : mResult.getSize();
    //
    mMediaRecorder.setVideoSize(size.getWidth(), size.getHeight());

---
The rest of the parameters are all exactly the same.

Here is a [link](https://sourceforge.net/p/opencamera/code/ci/master/tree/app/src/main/java/net/sourceforge/opencamera/preview/VideoProfile.java) to the class I'm referring to.

HBiSoft, 2019-05-27:
Not sure if it's worth mentioning that I'm starting a animation as soon as the user press the record button, as shown below:

    //Create animation
    Animation anim = new AlphaAnimation(0.0f, 1.0f);
    anim.setDuration(500);
    anim.setStartOffset(20);
    anim.setRepeatMode(Animation.REVERSE);
    anim.setRepeatCount(Animation.INFINITE);

    //start record button OnClickListener
    startRec.setOnClickListener(new View.OnClickListener() {
        @Override
        public void onClick(View v) {
            //set start record button invisible
            startRec.setVisibility(View.GONE);
            //set stop button visible
            stopRec.setVisibility(View.VISIBLE);
            //start recording
            startCaptureVideo();
            //start the stop button animation
            stopRec.startAnimation(anim);

        }

    });

natario1, 2019-05-27:
You can set Audio to OFF and see what happens... Make sure you are not changing anything else like the parameters I have listed above, VideoCodec...

HBiSoft, 2019-05-27:
@natario1 

I tested by changing all the parameters. 

This is all the parameters I've tested (individually):

- Disabling audio:
`cameraView.setAudio(Audio.OFF);`

- Change codec to 264:
`cameraView.setVideoCodec(VideoCodec.H_264);`

- Change codec to 263:
cameraView.setVideoCodec(VideoCodec.H_263);`

- Changing the video bitrate:
`cameraView.setVideoBitRate(3000000);`

---
The only successful test was changing the codec to:

    cameraView.setVideoCodec(VideoCodec.H_263);

So I assume the issue is with setting: 

    case DEVICE_DEFAULT: mMediaRecorder.setVideoEncoder(mProfile.videoCodec); break;

For some reason it throws an error when I don't set the video codec and it uses the default `DEVICE_DEFAULT` or 0.

I asked the user to send me a demo video that has been recorded with the devices camera, here is the video if you want to check:

https://drive.google.com/file/d/1CZ1SPxBtSChI3pRJK9WRbu4i464JYDK8/view?usp=sharing

Strangely, you will notice that the video he sent me (above), which is recorded with the devices camera, shows that the video codec is H264, but using `VideoCodec.H_264` causes the crash as well.

HBiSoft, 2019-05-28:
@natario1 

Should - 

    case DEVICE_DEFAULT: mMediaRecorder.setVideoEncoder(mProfile.videoCodec); break;

not be - 

    case DEVICE_DEFAULT: mMediaRecorder.setVideoEncoder(MediaRecorder.VideoEncoder.DEFAULT); break;

According to the [docs](https://developer.android.com/reference/android/media/MediaRecorder.VideoEncoder.html), it is true.

Also, CamcorderProfile in the [docs](https://developer.android.com/reference/android/media/CamcorderProfile#videoCodec).

---
I'm not sure, I'm just asking because the default currently is the cause of the crash.

wider2, 2019-05-29:
I have the same bug with Reason 5
my situation is
I can make some video records and after some number of successful attempt - camera shows the error
com.otaliastudios.cameraview.CameraView RuntimeException start failed

all right we caught the exception. Can we somehow restart the camera on the fly?

HBiSoft, 2019-05-29:
@wider2 We get the same error, but we don't have the same issue, this issue is that the wrong video encoder is selected.

In your case, I think you can restart the camera when the error occurs by calling:

    cameraView.close();
    cameraView.open();

HBiSoft, 2019-05-29:
@natario1 

I did another test, in my activity I get the video codec from `CamcorderProfile`, as shown below:

    //In onCreate
    CamcorderProfile camcorderProfile = CamcorderProfile.get(CamcorderProfile.QUALITY_HIGH);
    String profileInfo = getVideoCodec(camcorderProfile.videoCodec);

    //Then I check the codec and set it to CameraView
    if (profileInfo.equals("H264")){
        cameraView.setVideoCodec(VideoCodec.H_264);
    }
    if (profileInfo.equals("H263")){
        cameraView.setVideoCodec(VideoCodec.H_263);
    }
    if (profileInfo.equals("DEFAULT")){
        cameraView.setVideoCodec(VideoCodec.DEVICE_DEFAULT);
    }

    ////

    private String getVideoCodec(int videoCodec){
        switch(videoCodec){
            case MediaRecorder.VideoEncoder.H263:
                return "H263";
            case MediaRecorder.VideoEncoder.H264:
                return "H264";
            case MediaRecorder.VideoEncoder.MPEG_4_SP:
                return "MPEG_4_SP";
            case MediaRecorder.VideoEncoder.DEFAULT:
                return "DEFAULT";
            default:
                return "unknown";
        }
    }

The device that is experiencing the issue returns `MediaRecorder.VideoEncoder.DEFAULT`, so I then set the video codec to `cameraView.setVideoCodec(VideoCodec.DEVICE_DEFAULT);`, and the issue is still there.

The only way I'm able to get video recording to working on this device is to set the codec too H263 - `cameraView.setVideoCodec(VideoCodec.H_263);`

wider2, 2019-05-29:
my problem is quite similar like HBiSoft has

if camera return exception... somehow he remember about it and no possibility to refresh the state

this does not help

                @Override
                public void onCameraError(@NonNull CameraException exception) {
                    super.onCameraError(exception);
                    Log.e("onCameraError", "error: " + ((exception != null) ? exception.getMessage() : ""));

                    cameraView.close();
                    cameraView.open();

HBiSoft, 2019-05-29:
@wider2 Can you reproduce the error on the demo app?

HBiSoft, 2019-05-30:
@natario1 Another test...

I sent the user that is experiencing the issue the demo app.

He is getting the following crash when selecting `device default` and `h 264`:

    Got CameraException # 5 

Only when he selects `h 263` does he not get the error.

HBiSoft, 2019-05-30:
It seems like this issue was fixed and then re-introduced, perhaps when creating v2.

HBiSoft, 2019-06-02:
I will be implementing OpenCamera into my application for now, I can't have users experience crashes anymore. I will definitely keep an eye on this issue and return to this library when it gets fixed.

I will keep this issue open as the error is still present.

I have also asked a question about this on SO, here is the link - https://stackoverflow.com/questions/56342378/camcorderprofile-videocodec-returning-wrong-value

natario1, 2019-06-03:
It's not that the encoder is wrong... But rather the selected encoder from the profile (which probably is h264) is rejecting some of the other parameters. Meaning, it is not compatible with the selected size/bitrate/other stuff. 

There's no easy way to fix this without owning one of the devices. CameraView could choose a "better" profile (instead of always choosing QUALITY_HIGH) based on the incoming video size. 

Maybe, given your size, this "better" profile would come with a different videoCodec (h263?) or different parameters that are accepted by h264. But that's a shot in the dark.

I don't have much time at the moment but it shouldn't be so difficult (you can always donate to the project to accelerate. Sorry!).

HBiSoft, 2019-06-03:
@natario1  I donated :)

HBiSoft, 2019-06-03:
@natario1 I also think you are right when saying:

>CameraView could choose a "better" profile (instead of always choosing QUALITY_HIGH) based on the incoming video size.

Referring back to the OpenCamera implementation, you will see the [initialiseVideoSizes](https://sourceforge.net/p/opencamera/code/ci/master/tree/app/src/main/java/net/sourceforge/opencamera/preview/Preview.java#l2834) being called. Also, setting the profile in [initialiseVideoQuality](https://sourceforge.net/p/opencamera/code/ci/master/tree/app/src/main/java/net/sourceforge/opencamera/preview/Preview.java#l2843) based on the video size.

I THINK the issue might be that `CamcorderProfile.QUALITY...` should be set based on the video size.

natario1, 2019-06-09:
@HBiSoft are you able to test the cam-profiles branch? #477

natario1, 2019-06-12:
Merged into master now.

HBiSoft, 2019-06-12:
@natario1 Sorry for the late response.

I implemented the new master branch into my demo project and asked the user if he can test again, I will let you know once he has tested. Thank you.

HBiSoft, 2019-06-12:
@natario1 

Still the same error as last time:

    exception.getReason() started -----
    5
    exception.getReason() ended ------

    exception.toString() started ------
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: start failed.
    exception.toString() ended ------

    exception.getMessage() started ------
    java.lang.RuntimeException: start failed.
    exception.getMessage() ended ------

    exception.getCause().toString() started ------
    java.lang.RuntimeException: start failed.
    exception.getCause().toString() ended ------

    exception.getCause().getMessage() started ------
    start failed.
    exception.getCause().getMessage() ended ------

natario1, 2019-08-21:
I'll treat this as fixed by #545.

If not fixed, at least it should crash with more useful messages. Either way, there's nothing more I can do here. Not sure if you want to test that out @HBiSoft . Thanks anyway!

HBiSoft, 2019-09-03:
@natario1 
Sorry for the late reply, I would firstly like to compliment you on the progress you have made with the library since the last time I had a look at it :ok_hand:

Unfortunately I will not be able to perform the test, I do not have contact with that user anymore. I will implement the library into my application and see if the issue arises again.

I do have one question, If you don't mind - Are you planning on implementing the ability to set the frame rate, depending on what the device supports, when recording a video? (I should perhaps have asked this in a new issue/question).

lucazin, 2019-10-01:
I have this kind of error when i open demo and change to frontal camera and START recording.

`Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x6300750064006f in tid 22635 (VideoEncoder), pid 20068

2019-10-01 10:41:44.168 22656-22656/? A/DEBUG: Build fingerprint: 'samsung/a70qxtc/a70q:9/PPR1.180610.011/A705MNXXU3ASG7:user/release-keys'
2019-10-01 10:41:44.168 22656-22656/? A/DEBUG: Revision: '12'
2019-10-01 10:41:44.168 22656-22656/? A/DEBUG: ABI: 'arm64'
2019-10-01 10:41:44.168 22656-22656/? A/DEBUG: pid: 20068, tid: 22635, name: VideoEncoder  >>> br.com.oneplace <<<

2019-10-01 10:41:44.303 22656-22656/? A/DEBUG: backtrace:
2019-10-01 10:41:44.303 22656-22656/? A/DEBUG:     #00 pc 000000000026a498  /vendor/lib64/egl/libGLESv2_adreno.so (EsxBltLib::CopyHostPtrBltSw(EsxBltCopyHostPtr const*)+416)
2019-10-01 10:41:44.303 22656-22656/? A/DEBUG:     #01 pc 0000000000261ff0  /vendor/lib64/egl/libGLESv2_adreno.so (EsxResource::UpdateSubResourceDirect(EsxContext*, unsigned int, int, EsxBox const*, EsxSurfaceAccessDesc const*, int)+1112)
2019-10-01 10:41:44.303 22656-22656/? A/DEBUG:     #02 pc 0000000000262d48  /vendor/lib64/egl/libGLESv2_adreno.so (EsxTextureObject::PerformShadowCopy(EsxContext*)+608)
2019-10-01 10:41:44.303 22656-22656/? A/DEBUG:     #03 pc 000000000024fa4c  /vendor/lib64/egl/libGLESv2_adreno.so (EsxContext::ValidateProgramTexUnitBoundObjs(EsxProgram*, int)+724)
2019-10-01 10:41:44.303 22656-22656/? A/DEBUG:     #04 pc 000000000024c648  /vendor/lib64/egl/libGLESv2_adreno.so (EsxContext::ValidateGfxState(EsxDrawDescriptor const*)+520)
2019-10-01 10:41:44.303 22656-22656/? A/DEBUG:     #05 pc 00000000002a4350  /vendor/lib64/egl/libGLESv2_adreno.so (EsxContext::DrawArraysInstanced(EsxPrimType, int, unsigned int, unsigned int)+200)
2019-10-01 10:41:44.303 22656-22656/? A/DEBUG:     #06 pc 0000000000441a28  /system/framework/arm64/boot-framework.oat (offset 0x41e000) (android.opengl.GLES10.glDrawArrays [DEDUPED]+168)
2019-10-01 10:41:44.303 22656-22656/? A/DEBUG:     #07 pc 00000000000b91c8  /dev/ashmem/dalvik-jit-code-cache_20068_20068 (deleted) (com.otaliastudios.cameraview.filter.BaseFilter.onDraw+56)
2019-10-01 10:41:44.303 22656-22656/? A/DEBUG:     #08 pc 00000000000ba31c  /dev/ashmem/dalvik-jit-code-cache_20068_20068 (deleted) (com.otaliastudios.cameraview.filter.BaseFilter.draw+220)
2019-10-01 10:41:44.303 22656-22656/? A/DEBUG:     #09 pc 00000000000a4368  /dev/ashmem/dalvik-jit-code-cache_20068_20068 (deleted) (com.otaliastudios.cameraview.internal.egl.EglViewport.drawFrame+296)
2019-10-01 10:41:44.303 22656-22656/? A/DEBUG:     #10 pc 00000000000bfe70  /dev/ashmem/dalvik-jit-code-cache_20068_20068 (deleted) (com.otaliastudios.cameraview.video.encoding.TextureMediaEncoder.onFrame+2288)
2019-10-01 10:41:44.303 22656-22656/? A/DEBUG:     #11 pc 00000000000ef078  /dev/ashmem/dalvik-jit-code-cache_20068_20068 (deleted) (com.otaliastudios.cameraview.video.encoding.TextureMediaEncoder.onEvent+296)
2019-10-01 10:41:44.303 22656-22656/? A/DEBUG:     #12 pc 00000000000c2054  /dev/ashmem/dalvik-jit-code-cache_20068_20068 (deleted) (com.otaliastudios.cameraview.video.encoding.MediaEncoder$3.run+324)
2019-10-01 10:41:44.303 22656-22656/? A/DEBUG:     #13 pc 00000000000a19d0  /dev/ashmem/dalvik-jit-code-cache_20068_20068 (deleted) (android.os.Handler.handleCallback+64)
2019-10-01 10:41:44.303 22656-22656/? A/DEBUG:     #14 pc 00000000000a20ac  /dev/ashmem/dalvik-jit-code-cache_20068_20068 (deleted) (android.os.Handler.dispatchMessage+60)
2019-10-01 10:41:44.303 22656-22656/? A/DEBUG:     #15 pc 0000000000087f70  /dev/ashmem/dalvik-jit-code-cache_20068_20068 (deleted) (android.os.Looper.loop+1200)
2019-10-01 10:41:44.303 22656-22656/? A/DEBUG:     #16 pc 000000000055964c  /system/lib64/libart.so (art_quick_invoke_static_stub+604)
2019-10-01 10:41:44.303 22656-22656/? A/DEBUG:     #17 pc 00000000000d02e8  /system/lib64/libart.so (art::ArtMethod::Invoke(art::Thread*, unsigned int*, unsigned int, art::JValue*, char const*)+232)
2019-10-01 10:41:44.303 22656-22656/? A/DEBUG:     #18 pc 0000000000280cc0  /system/lib64/libart.so (art::interpreter::ArtInterpreterToCompiledCodeBridge(art::Thread*, art::ArtMethod*, art::ShadowFrame*, unsigned short, art::JValue*)+344)
2019-10-01 10:41:44.303 22656-22656/? A/DEBUG:     #19 pc 000000000027acc8  /system/lib64/libart.so (bool art::interpreter::DoCall<false, false>(art::ArtMethod*, art::Thread*, art::ShadowFrame&, art::Instruction const*, unsigned short, art::JValue*)+968)
2019-10-01 10:41:44.304 22656-22656/? A/DEBUG:     #20 pc 00000000005298e0  /system/lib64/libart.so (MterpInvokeStatic+204)
2019-10-01 10:41:44.304 22656-22656/? A/DEBUG:     #21 pc 000000000054bb94  /system/lib64/libart.so (ExecuteMterpImpl+14612)
2019-10-01 10:41:44.304 22656-22656/? A/DEBUG:     #22 pc 0000000000c77420  /system/framework/boot-framework.vdex (android.os.HandlerThread.run+56)
2019-10-01 10:41:44.304 22656-22656/? A/DEBUG:     #23 pc 000000000025491c  /system/lib64/libart.so (_ZN3art11interpreterL7ExecuteEPNS_6ThreadERKNS_20CodeItemDataAccessorERNS_11ShadowFrameENS_6JValueEb.llvm.1823628367+488)
2019-10-01 10:41:44.304 22656-22656/? A/DEBUG:     #24 pc 0000000000518c70  /system/lib64/libart.so (artQuickToInterpreterBridge+1020)
2019-10-01 10:41:44.304 22656-22656/? A/DEBUG:     #25 pc 00000000005624fc  /system/lib64/libart.so (art_quick_to_interpreter_bridge+92)
[... cut ...]
2019-10-01 10:41:44.304 22656-22656/? A/DEBUG:     #29 pc 000000000045fe0c  /system/lib64/libart.so (art::InvokeVirtualOrInterfaceWithJValues(art::ScopedObjectAccessAlreadyRunnable const&, _jobject*, _jmethodID*, jvalue*)+424)
2019-10-01 10:41:44.304 22656-22656/? A/DEBUG:     #30 pc 000000000048b22c  /system/lib64/libart.so (art::Thread::CreateCallback(void*)+1120)
2019-10-01 10:41:44.304 22656-22656/? A/DEBUG:     #31 pc 0000000000091c3c  /system/lib64/libc.so (__pthread_start(void*)+36)
2019-10-01 10:41:44.304 22656-22656/? A/DEBUG:     #32 pc 0000000000024200  /system/lib64/libc.so (__start_thread+68)`

natario1, 2019-10-05:
@HBiSoft I believe this is fixed in #602 and related PR #617 . The only thing we had to do is to avoid choosing video sizes larger than `CamcorderProfile.QUALITY_HIGH`, even if `Camera.Parameters.getSupportedVideoSizes()` returns one of them.

It's not clear whether we should trust the camera video sizes, the camcorder profiles, ... . Seems to be device-dependent nightmare as usual. 

I'd be curious to know if OpenCamera does something similar, but I find it very hard to go through their code style. Let me know if you have this information!

About your frame rate question, it's not easy, but we might implement it one day.

closed 2019-10-07 by natario1

HBiSoft, 2019-11-06:
@natario1 Thanks for letting me know. This issue seems to be fixed. I had a quick look at the OpenCamera project and he uses the following method the set the profile:

```java
private void initialiseVideoQuality() {
    int cameraId = camera_controller.getCameraId();
    List<Integer> profiles = new ArrayList<>();
    List<VideoQualityHandler.Dimension2D> dimensions = new ArrayList<>();

     if( CamcorderProfile.hasProfile(cameraId, CamcorderProfile.QUALITY_HIGH) ) {
        CamcorderProfile profile = CamcorderProfile.get(cameraId, CamcorderProfile.QUALITY_HIGH);
        profiles.add(CamcorderProfile.QUALITY_HIGH);
        dimensions.add(new VideoQualityHandler.Dimension2D(profile.videoFrameWidth, profile.videoFrameHeight));
    }
    if( Build.VERSION.SDK_INT >= Build.VERSION_CODES.LOLLIPOP ) {
        if( CamcorderProfile.hasProfile(cameraId, CamcorderProfile.QUALITY_2160P) ) {
            CamcorderProfile profile = CamcorderProfile.get(cameraId, CamcorderProfile.QUALITY_2160P);
            profiles.add(CamcorderProfile.QUALITY_2160P);
            dimensions.add(new VideoQualityHandler.Dimension2D(profile.videoFrameWidth, profile.videoFrameHeight));
        }
    }
    if( CamcorderProfile.hasProfile(cameraId, CamcorderProfile.QUALITY_1080P) ) {
        CamcorderProfile profile = CamcorderProfile.get(cameraId, CamcorderProfile.QUALITY_1080P);
        profiles.add(CamcorderProfile.QUALITY_1080P);
        dimensions.add(new VideoQualityHandler.Dimension2D(profile.videoFrameWidth, profile.videoFrameHeight));
    }
    if( CamcorderProfile.hasProfile(cameraId, CamcorderProfile.QUALITY_720P) ) {
        CamcorderProfile profile = CamcorderProfile.get(cameraId, CamcorderProfile.QUALITY_720P);
        profiles.add(CamcorderProfile.QUALITY_720P);
        dimensions.add(new VideoQualityHandler.Dimension2D(profile.videoFrameWidth, profile.videoFrameHeight));
    }
    if( CamcorderProfile.hasProfile(cameraId, CamcorderProfile.QUALITY_480P) ) {
        CamcorderProfile profile = CamcorderProfile.get(cameraId, CamcorderProfile.QUALITY_480P);
        profiles.add(CamcorderProfile.QUALITY_480P);
        dimensions.add(new VideoQualityHandler.Dimension2D(profile.videoFrameWidth, profile.videoFrameHeight));
    }
    if( CamcorderProfile.hasProfile(cameraId, CamcorderProfile.QUALITY_CIF) ) {
        CamcorderProfile profile = CamcorderProfile.get(cameraId, CamcorderProfile.QUALITY_CIF);
        profiles.add(CamcorderProfile.QUALITY_CIF);
        dimensions.add(new VideoQualityHandler.Dimension2D(profile.videoFrameWidth, profile.videoFrameHeight));
    }
    if( CamcorderProfile.hasProfile(cameraId, CamcorderProfile.QUALITY_QVGA) ) {
        CamcorderProfile profile = CamcorderProfile.get(cameraId, CamcorderProfile.QUALITY_QVGA);
        profiles.add(CamcorderProfile.QUALITY_QVGA);
        dimensions.add(new VideoQualityHandler.Dimension2D(profile.videoFrameWidth, profile.videoFrameHeight));
    }
    if( CamcorderProfile.hasProfile(cameraId, CamcorderProfile.QUALITY_QCIF) ) {
        CamcorderProfile profile = CamcorderProfile.get(cameraId, CamcorderProfile.QUALITY_QCIF);
        profiles.add(CamcorderProfile.QUALITY_QCIF);
        dimensions.add(new VideoQualityHandler.Dimension2D(profile.videoFrameWidth, profile.videoFrameHeight));
    }
    if( CamcorderProfile.hasProfile(cameraId, CamcorderProfile.QUALITY_LOW) ) {
        CamcorderProfile profile = CamcorderProfile.get(cameraId, CamcorderProfile.QUALITY_LOW);
        profiles.add(CamcorderProfile.QUALITY_LOW);
        dimensions.add(new VideoQualityHandler.Dimension2D(profile.videoFrameWidth, profile.videoFrameHeight));
    }
    this.video_quality_handler.initialiseVideoQualityFromProfiles(profiles, dimensions);
}
```

The `initialiseVideoQualityFromProfiles` points to [this](https://sourceforge.net/p/opencamera/code/ci/master/tree/app/src/main/java/net/sourceforge/opencamera/preview/VideoQualityHandler.java#l60) class. The profile is also selected based on the fps provided. He gets the supported video sizes by passing the fps and checking if it is supported.

To get the max supported frame rate, one can use the following:

```java
CameraManager manager = (CameraManager)activity.getSystemService(Context.CAMERA_SERVICE);
String cameraId = manager.getCameraIdList()[0];
CameraCharacteristics characteristics = manager.getCameraCharacteristics(cameraId);
StreamConfigurationMap map = characteristics.get(CameraCharacteristics.SCALER_STREAM_CONFIGURATION_MAP);
Range<Integer>[] fpsRange = map.getHighSpeedVideoFpsRanges();
```

and if the device is not `FULL_LEVEL`, the max frame rate can be derived like this:

```java
CameraManager manager = (CameraManager)activity.getSystemService(Context.CAMERA_SERVICE);
String cameraId = manager.getCameraIdList()[0];
CameraCharacteristics characteristics = manager.getCameraCharacteristics(cameraId);
StreamConfigurationMap map = characteristics.get(CameraCharacteristics.SCALER_STREAM_CONFIGURATION_MAP);
Range<Integer>[] fpsRange = characteristics.get(CameraCharacteristics.CONTROL_AE_AVAILABLE_TARGET_FPS_RANGES);
```

It will really be a great addition to this project to be able to adjust the frame rate. This has been a highly requested feature in my application as well.

PR #617 (2019-10-07): takeVideo improvements
Fixes #602 
Fixes #467