natario1/CameraView #469: takeVideoSnapshot is not working properly in demo app in LG G7 ThinQ

sudhakar434, 2019-05-24:
### Describe the bug
In the official demo app, video snapshot is getting struck at preview screen at first frame but the audio is playing

- CameraView version: v2.0.0-beta04
- Reproducible in official demo app: yes
- Device / Android version: LG G7 ThinQ, API 26

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Go to Demo app
2. Click on settings icon and change to video mode
3. Now click on video snap shot record option 
4. After 5 sec of video record, in the preview screen video is struck at first frame but audio is playing

### Screenshots
[image]
[image]
[image]

In the first screenshot, video is recorded and in the second screen video will get stuck at first frame but audio will be playing.
Please help me with this issue. Thank you very much for providing the wonderful camera library.

dummy-amit, 2019-05-25:
@natario1 same error for me as well, could you please shed some light on this. thanks.

natario1, 2019-05-25:
You can follow #449

closed 2019-05-25 by natario1