natario1/CameraView #470: ImageFormat Exception

brnogz, 2019-05-24:
### Describe the bug
A clear and concise description of what the bug is.

- CameraView version: *version number*
- Reproducible in official demo app: *yes/no*
- Device / Android version: *Pixel, API 28*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Go to '...'
2. Click on '....'
3. See error

### Expected behavior
A clear and concise description of what you expected to happen.

### Screenshots
If applicable, add screenshots to help explain your problem.

### Logs
Use `CameraLogger.setLogLevel(LEVEL_VERBOSE)` to see all logs into LogCat.
Use `CameraLogger.registerLogger()` to export to file or crash reporting service.

### APK
Link to a Github repo where the bug is reproducible.

HBiSoft, 2019-06-01:
@brnogz I don't think it's related to the library, have a look ate this question/answer - https://stackoverflow.com/questions/11511522/byte-array-image-to-bitmap

brnogz, 2019-06-11:
@HBiSoft I don't do any image conversion or processing.  Logs include only cameraview and android classes and YuvImage used in cameraview. Because of this, i create this issue but if it is a configuration error of mine, could you help me to solve?

brnogz, 2019-07-02:
what a nice way to manage issues

natario1, 2019-07-11:
This should be fixed now.

closed 2019-07-11 by natario1