natario1/CameraView #473: How can I save a captured frame as an image file?

ankitivo, 2019-06-01:
Hi,
I have a very general requirement where I want to process a frame and then save that frame as an image file based on the result of my processing.

I am able to successfully process a frame byte[] but facing an issue while saving the frame to an image file.

I am using below code to convert frame byte[] into a bitmap:

`Size size = SystemServiceUtils.getWindowRealSize(context);
        Bitmap bitmap = CameraUtils.decodeBitmap(data.array(), size.getMaxOfWidthHeight(), size.getMaxOfWidthHeight(),
                new BitmapFactory.Options());`

but the converted bitmap is always null. I see the following error in the log line:

` D/skia: --- Failed to create image decoder with message 'unimplemented'`

I am using the 2.0.0-beta05 version of this library.

natario1, 2019-08-19:
That's because frames come in `ImageFormat.NV21` (as you can see from frame.getFormat()) so they are not JPEGs. You should find some other way of visualizing, maybe using the `YuvImage` class.

closed 2019-08-19 by natario1

ankitivo, 2019-08-20:
Thanks, I did figure it out.