natario1/CameraView #476: A way to Pause/Freeze camera preview?

anthonynguyen194, 2019-06-07:
### How do I?
I've been reading through the docs and didn't see any mentions of being able to freeze/pause the frame of the camera view. 

All I want to do is have the user capture an image and freeze the preview afterwards. 

Is this possible with CamerView?

### Version used
CameraView exact version.

hanspeide, 2019-06-08:
@natario1 Could this be easy as calling `.stopPreview()` in `.takePicture()`, and then `.startPreview()` in the `onPictureResult` callback? I just tried, and it seems to work.

I can send a PR if you want. Would you then in `Camera1.java` prever to add a method `takePictureAndStopPreview()`, or `takePicture(boolean stopPreview)`?

hanspeide, 2019-06-08:
Alternatively methods `.stopPreview()` and `.startPreview()` can be exposed in `CameraView.java`. That might be a better solution?

anthonynguyen194, 2019-06-08:
I think exposing the .stopPreview() and .startPreview() method would be better rather than combining the takePicture and stopPreview functionality together.

natario1, 2019-06-09:
Thanks! I disagree with introducing a new camera state with the whole confusion it brings in, for just a few usages, when there's a super easy solution already - take the picture / picture snapshot, and set it as a drawable to an ImageView on top of CameraView so it's covered.

hanspeide, 2019-06-10:
@natario1 My issue here is the (obvious) delay between calling `.takePicture()` and receiving the result in `onPictureTaken()`. For UX reasons I would like to freeze the frame preview once `.takePicture()` is called, to prevent the user from thinking the camera is still active after taking a picture. 

Were you think about taking the picture and snapshot sequentially, and then setting the snapshot as the `ImageView` drawable? From what I can tell it doesn't seem to work. An alternative would be to first call `takeSnapshot()`, then in `onPictureTaken()` check for `result.isSnapshot` if it is a snapshot call `takePicture()`. But, that doesn't feel very elegant either.

natario1, 2019-06-10:
I don't remember atm if you can call both together in new version, but if not, yes, snapshot first! The snapshot latency can be extremely low.

For "elegance" I would ignore snapshots in your main callback, and use a one-time callback just for the snapshot.

But keep an eye on here, at some point we can make this possible in other ways (let you subclass some component so you can avoid drawing, for example), I just don't like the idea of a frozen state.

natario1, 2019-06-10:
Another option is use the FrameProcessor to get the latest frame when shooting, that's the faster thing but it doesn't come in JPEG, prob. decoding it takes more time but I don't really know.

natario1, 2019-06-10:
What I can accept as a PR is give the developer a boolean option, that if true, freezes the preview during the time between take and onTaken.

Tecnhically this should be done in the Preview classes (avoid drawing) rather than tearing the preview down with stopPreview.