natario1/CameraView #487: Spot metering

cvb941, 2019-07-01:
Is there any way of changing the metering strategy, with modes that can be found on usual cameras? Or just setting the exposure areas, [as found in the camera api](https://developer.android.com/reference/android/hardware/Camera.Parameters.html#setMeteringAreas(java.util.List%3Candroid.hardware.Camera.Area%3E))? 

We use the manual exposure correction for now, but it's not satisfactory. Thank you for your answers.

natario1, 2019-07-04:
Can you list the Metering strategies that you would like and how each one translates into metering areas / weights?

cvb941, 2019-07-04:
If you'd like a predefined list of them, together with [this wikipedia page](https://en.wikipedia.org/wiki/Metering_mode) I would say:

- Default - equivalent to null on the setMeteringAreas(), use default device implementation
- Spot metering - metering based on the center 5% area 
- Center weighted average metering - metering based on the center 70% area 
- Average metering - metering based on the full image

I guess the weights can be 1000 in the areas, as only the area should matter to the metering. We don't want to meter the rest of the picture, so either leave them as be, or set them to 0, if the former doesn't work.