natario1/CameraView #489: Taken photo is bigger than the preview

luangs7, 2019-07-04:
### How do I?
I'm facing a problem using a Samsung J1 device here.
But the image taken is bigger than the preview one.

This is the preview that I got:
[image]

and this is the final bitmap result:

[image]

I'm calling camera view with this attributes:
 <com.otaliastudios.cameraview.CameraView
            android:id="@+id/camera"
            app:cameraGesturePinch="zoom"
            app:cameraFlash="auto"
            app:cameraAutoFocusResetDelay="0"
            app:cameraGestureTap="focusWithMarker"
            android:keepScreenOn="true"
            app:cameraPreview="glSurface"
            app:cameraPictureSizeSmallest="true"
            android:adjustViewBounds="true"
            app:cameraGestureScrollHorizontal="exposureCorrection"
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            app:layout_constraintBottom_toBottomOf="parent"
            app:layout_constraintEnd_toEndOf="parent"
            app:layout_constraintStart_toStartOf="parent"
            app:layout_constraintTop_toTopOf="parent" />

I already tried to use cameraPictureSizeAspectRatio with 4:3 but no success with it.
### Version used
2.0.0-beta05

natario1, 2019-07-04:
You can either use takePictureSnapshot(), OR leave one of the CameraView dimensions to wrap_content .

luangs7, 2019-07-04:
Thanks for your help, I'll try it when I return to the office.

Looking at the images, only the height is diff between the final bitmap and the preview.
Do you think that is a aspect ratio problem because of the portrait orientation?

natario1, 2019-07-04:
Your screen aspect ratio is not supported by your device, so since you use match_parent for both dimensions, CameraView trims the exceeding top&bottom during the preview.

luangs7, 2019-07-04:
I get it.
Work as expect using takePictureSnapshot.
Thanks mate!

closed 2019-07-04 by natario1