natario1/CameraView #491: App crash when tap to focus

luangs7, 2019-07-05:
### Describe the bug
App crash with focus on tap.

- CameraView version: 2.0.0-beta05
- Device / Android version: *Samsung Gran Prime Duos, API 22*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Open camera
2. Tap on preview to focus, with app:cameraGestureTap="focusWithMarker"
3. See error

### Expected behavior
If focus does not support, dont crash the app.

### Logs
java.lang.RuntimeException: setParameters failed
        at android.hardware.Camera.native_setParameters(Native Method)
        at android.hardware.Camera.setParameters(Camera.java:1946)
        at com.otaliastudios.cameraview.Camera1$19.run(Camera1.java:883)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:61)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:135)
        at android.os.HandlerThread.run(HandlerThread.java:61)

luangs7, 2019-07-11:
Any expectation to correct this one?

natario1, 2019-07-11:
Is this your own device? @luangs7

luangs7, 2019-07-22:
Yes, it is. @natario1. Happening again with beta06 version

natario1, 2019-07-24:
Can you download the latest demo app and do some debugging to see which line is causing this. You should modify activity_camera.xml to use Camera1 instead of Camera2 and then you could do some tests... For example:

1. remove line 792 only (Camera1Engine.java)
2. remove line 793 only
3. remove 792 and 793
4. remove 794 only

In which of these scenarios does it crash?

closed 2019-08-20 by stale[bot]