natario1/CameraView #503: Runtime crash

JirkaKrivanek, 2019-07-17:
V2 does not work to me at all, crashes on runtime on some missing component (build is fine):

Caused by: java.lang.NoClassDefFoundError: Failed resolution of: Lcom/google/android/gms/tasks/Tasks;
at com.otaliastudios.cameraview.engine.CameraEngineStep.<init>(CameraEngineStep.java:57)
at com.otaliastudios.cameraview.engine.CameraEngine.<init>(CameraEngine.java:194)
at com.otaliastudios.cameraview.engine.Camera1Engine.<init>(Camera1Engine.java:69)
at com.otaliastudios.cameraview.CameraView.instantiateCameraEngine(CameraView.java:263)
at com.otaliastudios.cameraview.CameraView.doInstantiateEngine(CameraView.java:235)
at com.otaliastudios.cameraview.CameraView.init(CameraView.java:191)

zack-barakat, 2019-07-17:
I had the same issue, I solved it temporarly by adding CamerViea dependancies in my app build.gradle

`   implementation 'androidx.exifinterface:exifinterface:1.0.0'
    implementation 'androidx.lifecycle:lifecycle-common:2.1.0-alpha01'
    implementation 'com.google.android.gms:play-services-tasks:17.0.0'
`

natario1, 2019-07-17:
If you have missing dependencies you are either pulling the library in a unordinary way or you are using the AAR for some reason?

natario1, 2019-07-22:
Please subscribe to #516

closed 2019-07-22 by natario1