natario1/CameraView #504: remove gms dependencies

fanxxxx, 2019-07-17:
### Problem to be addressed
CameraEngine use com.google.android.gms.tasks.Task **For convenience** 
but at china, we don't support gms, It's not very convenient.
### Describe the solution you'd like to
remove gms dependencies

natario1, 2019-07-17:
What do you mean? What happens if you run the app?

The Task dependency is just an ordinary library, it is not really a Google Play Services lib (e.g. requires no play services plugin, configuration, ...).

maylamayana, 2019-07-21:
cameraview:2.0.0-rc1 

E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.example.mydic, PID: 2313
    java.lang.RuntimeException: Unable to start activity ComponentInfo{com.example.mydic/com.example.mydic.MainActivity}: android.view.InflateException: Binary XML file line #9: Binary XML file line #9: Error inflating class com.otaliastudios.cameraview.CameraView
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2927)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2988)
        at android.app.ActivityThread.-wrap14(ActivityThread.java)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1631)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:154)
        at android.app.ActivityThread.main(ActivityThread.java:6682)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1520)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1410)
     Caused by: android.view.InflateException: Binary XML file line #9: Binary XML file line #9: Error inflating class com.otaliastudios.cameraview.CameraView
     Caused by: android.view.InflateException: Binary XML file line #9: Error inflating class com.otaliastudios.cameraview.CameraView
     Caused by: java.lang.reflect.InvocationTargetException
        at java.lang.reflect.Constructor.newInstance0(Native Method)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:430)
        at android.view.LayoutInflater.createView(LayoutInflater.java:652)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:794)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:734)
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:865)
        at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:828)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:525)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:427)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:378)
        at androidx.appcompat.app.AppCompatDelegateImpl.setContentView(AppCompatDelegateImpl.java:545)
        at androidx.appcompat.app.AppCompatActivity.setContentView(AppCompatActivity.java:161)
        at com.example.mydic.MainActivity.onCreate(MainActivity.java:119)
        at android.app.Activity.performCreate(Activity.java:6942)
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1126)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2880)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2988)
        at android.app.ActivityThread.-wrap14(ActivityThread.java)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1631)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:154)
        at android.app.ActivityThread.main(ActivityThread.java:6682)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1520)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1410)
     Caused by: java.lang.NoClassDefFoundError: Failed resolution of: Lcom/google/android/gms/tasks/Tasks;
        at com.otaliastudios.cameraview.engine.Step.<init>(Step.java:57)
        at com.otaliastudios.cameraview.engine.CameraEngine.<init>(CameraEngine.java:198)
        at com.otaliastudios.cameraview.engine.Camera1Engine.<init>(Camera1Engine.java:69)
        at com.otaliastudios.cameraview.CameraView.instantiateCameraEngine(CameraView.java:279)
        at com.otaliastudios.cameraview.CameraView.doInstantiateEngine(CameraView.java:247)
        at com.otaliastudios.cameraview.CameraView.init(CameraView.java:201)
        at com.otaliastudios.cameraview.CameraView.<init>(CameraView.java:149)
        at java.lang.reflect.Constructor.newInstance0(Native Method) 
        at java.lang.reflect.Constructor.newInstance(Constructor.java:430) 
        at android.view.LayoutInflater.createView(LayoutInflater.java:652) 
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:794) 
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:734) 
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:865) 
        at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:828) 
        at android.view.LayoutInflater.inflate(LayoutInflater.java:525) 
        at android.view.LayoutInflater.inflate(LayoutInflater.java:427) 
        at android.view.LayoutInflater.inflate(LayoutInflater.java:378) 
        at androidx.appcompat.app.AppCompatDelegateImpl.setContentView(AppCompatDelegateImpl.java:545) 
        at androidx.appcompat.app.AppCompatActivity.setContentView(AppCompatActivity.java:161) 
        at com.example.mydic.MainActivity.onCreate(MainActivity.java:119) 
        at android.app.Activity.performCreate(Activity.java:6942) 
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1126) 
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2880) 
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2988) 
        at android.app.ActivityThread.-wrap14(ActivityThread.java) 
[... cut ...] .apk!/lib/arm64-v8a, /data/app/com.example.mydic-1/split_lib_slice_7_apk.apk!/lib/arm64-v8a, /data/app/com.example.mydic-1/split_lib_slice_8_apk.apk!/lib/arm64-v8a, /data/app/com.example.mydic-1/split_lib_slice_9_apk.apk!/lib/arm64-v8a, /system/lib64, /vendor/lib64]]
        at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:380)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
        	... 32 more

maylamayana, 2019-07-21:
I solved it by adding CamerView dependancies in my app build.gradle:

implementation 'androidx.exifinterface:exifinterface:1.0.0'
    implementation 'androidx.lifecycle:lifecycle-common:2.2.0-alpha02'
    implementation 'com.google.android.gms:play-services-tasks:17.0.0'

fanxxxx, 2019-07-22:
> What do you mean? What happens if you run the app?
> 
> The Task dependency is just an ordinary library, it is not really a Google Play Services lib (e.g. requires no play services plugin, configuration, ...).

I add play-services-tasks to solved this. This dependency size is 27KB. 

For some app who didn't depend on gms, It's better to remove it

closed 2019-07-22 by natario1

abolfazlbzgh, 2020-02-19:
i use this and work
[Failed to resolve: play-services-tasks](https://stackoverflow.com/a/51232431/8748900)

    repositories {
        google()
        maven {
            url 'https://maven.google.com/'
        }
        jcenter()
        mavenCentral()
        maven {
            url 'https://maven.fabric.io/public'
        }
    }