natario1/CameraView #510: Camera zoom, when record video

borichellow, 2019-07-19:
- CameraView version: *v2.0.0-rc1*
- Reproducible in official demo app: *yes*
- Device / Android version: *Xiaomi Mi A1, Android 9*

#### Steps to reproduce the behavior in the demo app:
**(Engine is `camera2`)**
1. Go to `MainView`
2. Switch mode to Video 
3. Click `record video` button
4. See that video recording started, but picture is zoomed a bit (about x1.5)

#### Expected behavior
Video recording started, with original zoom

## Similar issue: but with Camera1

- CameraView version: *v2.0.0-rc1*, *v2.0.0-beta04*
- Reproducible in official demo app: *yes*
- Device / Android version: *Xiaomi Mi A1, Android 9*

#### Steps to reproduce the behavior in the demo app:
**(Engine is `camera1`)**
1. Go to `MainView`
2. Switch mode to Video 
3. See that preview is zoomed a bit (about x1.5)
4. Click `record video` button
5. wait till recording ended
5. See that result video doesn't have zoom issue (zoom is regular, not like it was while recording)

#### Expected behavior
Zoom doesn't change after changing mode to Video

kcstesting, 2019-07-24:
I have experience a same issue when start recording in video mode for iBall tablet. the screen will be zoomed so can you please fix this

marcin-adamczewski, 2019-08-07:
I encounter the same issue. Any fix for that?

natario1, 2019-08-07:
Can you post a screen recording of the issue? and specify if it's Camera1 or Camera2.

borichellow, 2019-08-22:
@natario1 here is the video https://www.dropbox.com/s/3guregjef6ax7r0/Untitled.mp4?dl=0
device is (Xiaomi Mi A1, Camera2 is used).
As you can see by "MackBook" label, when the recording started image zoomed a bit

natario1, 2019-08-22:
Thanks @borichellow that's interesting!

So you have this issue with both Camera1 and Camera2, with the only difference that in Camera1 it zooms as soon as you change mode, while in Camera2 it zooms when you start recording? Is this correct?

Does all this happen when you use `texture` as a preview surface?

borichellow, 2019-08-23:
@natario1 yes, in Camera1 it zooms as soon as I change mode and in Camera2 it zooms when I start recording.

I've just tried the newest demo-version: behaviour is the same, except one thing: in Camera1 i can't even record the video, get `CameraException#5` all the time.

For `texture` as a preview surface behaviour is the same.

natario1, 2019-08-23:
@borichellow but once it zooms, can you go back to the original zoom with the pinch-to-zoom gesture? Or is it an extra zoom that can not be undone? Try doing some zoom in/out there and see if you're able to restore the original zoom.

Also - could you try removing `params.setRecordingHint(getMode() == Mode.VIDEO);` from Camera1Engine (line 391 or so) and try again with camera 1? Let's ignore the crash for now since the issue happens before you start recording.

borichellow, 2019-08-26:
@natario1 zoom in/out - doesn't help, no option to get original zoom.

Removing `params.setRecordingHint(getMode() == Mode.VIDEO);` from Camera1Engine (line 391)  - looks like helped. It doesn't zoom anymore after switching to videoMode

Also i've tried to remove `params.setRecordingHint(getMode() == Mode.VIDEO);` AND adding 
```
if (params.isVideoStabilizationSupported()) {
    params.setVideoStabilization(false);
}
```
in that case camera1 also works good. I mean: no zoom after switching to videoMode, but still can't record video :)

natario1, 2019-08-26:
@borichellow have you tried leaving `params.setRecordingHint(getMode() == Mode.VIDEO)` untouched, as is, but adding the video stabilization stuff? We don't want to remove the recording hint.

About the video exception, if you keep helping, we'll fix both issues :-) I can't reproduce this on my devices. Thanks!

borichellow, 2019-08-26:
@natario1 unfortunately it always zooms preview in videoMode  with `params.setRecordingHint(getMode() == Mode.VIDEO)` :(

sure, i can help, lemme know what to do :)

natario1, 2019-08-26:
That sucks, we don't want to remove the recording hint. Let's see if we can fix this on Camera2 for now.

Can you make sure you use the Camera2 engine and modify the `Camera2Engine` class?
Go to the `applyAllParameters` method and add these lines at the top of the method:

```java
builder.set(CaptureRequest.CONTROL_VIDEO_STABILIZATION_MODE, CaptureRequest.CONTROL_VIDEO_STABILIZATION_MODE_OFF);
builder.set(CaptureRequest.LENS_OPTICAL_STABILIZATION_MODE, CaptureRequest.LENS_OPTICAL_STABILIZATION_MODE_OFF);
```

If this fixes the issue, could you see which line is fixing it ? The first, or the second, or do we have to add both?

borichellow, 2019-08-26:
@natario1 , tried both lines (separately and together), result is the same: zoom after starting record. 
Could be useful: my device has 2 cameras, so maybe it's not zoom, but switching between them.

natario1, 2019-08-26:
I'm out of ideas then @borichellow ! You might want to try apply these to each physical camera:

```java
for (String id : mCameraCharacteristics.getPhysicalCameraIds()) {
    builder.setPhysicalCameraKey(CaptureRequest.CONTROL_VIDEO_STABILIZATION_MODE, CaptureRequest.CONTROL_VIDEO_STABILIZATION_MODE_OFF, id);
    builder.setPhysicalCameraKey(CaptureRequest.LENS_OPTICAL_STABILIZATION_MODE, CaptureRequest.LENS_OPTICAL_STABILIZATION_MODE_OFF, id);
}
```

But other than this I don't know what else to say, I don't have a similar device.

To check your suggestion, you can check the size / contents of `mCameraCharacteristics.getPhysicalCameraIds()` before and after the bug and see if it changes.

borichellow, 2019-08-29:
@natario1 nothing helps :(

tried to check `mCameraCharacteristics.getPhysicalCameraIds()` but it's always empty. Is it actually should be empty? 

Another thing that could prove my "2 cameras theory" is that issue doesn't happen for front camera (there is only one front camera on my device)

natario1, 2019-09-04:
I don't think I can help further @borichellow .. without owning such device. If you find a solution feel free to post it here, otherwise I'll close this in some weeks. I also think it's related to having two cameras.

About the video issue please open a new issue with required information and we can take a look.

closed 2019-09-28 by natario1