natario1/CameraView #514: Video recording fails on Xiaomi Redmi 7

RAN3000, 2019-07-22:
### Describe the bug
A clear and concise description of what the bug is.

- CameraView version: commit f2350512f0193f3b2c80e6fa4deb324708e857d7 (first one after v2.0.0-rc1)
- Reproducible in official demo app: yes
- Device / Android version: Xiaomi Redmi 7, MIUI Global 10.3.1, API level 28

### To Reproduce
Just take a video (for the logs below I set the video duration to 2 seconds, but it affects every length).
The app does not crash but the video is recorded as a black background with the rotating watermark in purple.

### Screenshots
[image]
[image]

### Logs
```
2019-07-22 12:12:35.712 32339-32339/? I/cameraview.dem: Late-enabling -Xcheck:jni
2019-07-22 12:12:36.009 32339-32339/com.otaliastudios.cameraview.demo W/cameraview.dem: JIT profile information will not be recorded: profile file does not exits.
2019-07-22 12:12:36.011 32339-32339/com.otaliastudios.cameraview.demo I/chatty: uid=10247(com.otaliastudios.cameraview.demo) identical 10 lines
2019-07-22 12:12:36.011 32339-32339/com.otaliastudios.cameraview.demo W/cameraview.dem: JIT profile information will not be recorded: profile file does not exits.
2019-07-22 12:12:36.044 32339-32339/com.otaliastudios.cameraview.demo I/Perf: Connecting to perf service.
2019-07-22 12:12:36.063 32339-32339/com.otaliastudios.cameraview.demo I/InstantRun: starting instant run server: is main process
2019-07-22 12:12:36.285 32339-32339/com.otaliastudios.cameraview.demo W/cameraview.dem: Accessing hidden method Landroid/view/View;->computeFitSystemWindows(Landroid/graphics/Rect;Landroid/graphics/Rect;)Z (light greylist, reflection)
2019-07-22 12:12:36.286 32339-32339/com.otaliastudios.cameraview.demo W/cameraview.dem: Accessing hidden method Landroid/view/ViewGroup;->makeOptionalFitsSystemWindows()V (light greylist, reflection)
2019-07-22 12:12:36.361 32339-32339/com.otaliastudios.cameraview.demo E/libc: Access denied finding property "vendor.camera.aux.packagelist"
2019-07-22 12:12:36.352 32339-32339/com.otaliastudios.cameraview.demo W/cameraview.demo: type=1400 audit(0.0:232537): avc: denied { read } for name="u:object_r:vendor_camera_prop:s0" dev="tmpfs" ino=13672 scontext=u:r:untrusted_app:s0:c247,c256,c512,c768 tcontext=u:object_r:vendor_camera_prop:s0 tclass=file permissive=0
2019-07-22 12:12:36.364 32339-32339/com.otaliastudios.cameraview.demo E/libc: Access denied finding property "vendor.camera.aux.packagelist"
2019-07-22 12:12:36.352 32339-32339/com.otaliastudios.cameraview.demo W/cameraview.demo: type=1400 audit(0.0:232538): avc: denied { read } for name="u:object_r:vendor_camera_prop:s0" dev="tmpfs" ino=13672 scontext=u:r:untrusted_app:s0:c247,c256,c512,c768 tcontext=u:object_r:vendor_camera_prop:s0 tclass=file permissive=0
2019-07-22 12:12:36.521 32339-32339/com.otaliastudios.cameraview.demo W/cameraview.dem: Accessing hidden method Landroid/graphics/FontFamily;-><init>()V (light greylist, reflection)
2019-07-22 12:12:36.527 32339-32339/com.otaliastudios.cameraview.demo E/TypefaceCompatApi21Impl: java.lang.NoSuchMethodException
    java.lang.NoSuchMethodException: addFontWeightStyle [class java.lang.String, int, boolean]
        at java.lang.Class.getMethod(Class.java:2068)
        at java.lang.Class.getMethod(Class.java:1690)
        at androidx.core.graphics.TypefaceCompatApi21Impl.<clinit>(TypefaceCompatApi21Impl.java:74)
        at androidx.core.graphics.TypefaceCompat.<clinit>(TypefaceCompat.java:47)
        at androidx.core.graphics.TypefaceCompat.create(TypefaceCompat.java:190)
        at androidx.appcompat.widget.AppCompatTextView.setTypeface(AppCompatTextView.java:576)
        at android.widget.TextView.resolveStyleAndSetTypeface(TextView.java:2037)
        at android.widget.TextView.setTypefaceFromAttrs(TextView.java:2008)
        at android.widget.TextView.applyTextAppearance(TextView.java:3644)
        at android.widget.TextView.<init>(TextView.java:1498)
        at android.widget.TextView.<init>(TextView.java:869)
        at androidx.appcompat.widget.AppCompatTextView.<init>(AppCompatTextView.java:97)
        at androidx.appcompat.widget.AppCompatTextView.<init>(AppCompatTextView.java:93)
        at androidx.appcompat.app.AppCompatViewInflater.createTextView(AppCompatViewInflater.java:182)
        at androidx.appcompat.app.AppCompatViewInflater.createView(AppCompatViewInflater.java:103)
        at androidx.appcompat.app.AppCompatDelegateImpl.createView(AppCompatDelegateImpl.java:1267)
        at androidx.appcompat.app.AppCompatDelegateImpl.onCreateView(AppCompatDelegateImpl.java:1317)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:776)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:734)
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:867)
        at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:828)
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:870)
        at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:828)
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:870)
        at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:828)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:519)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:427)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:374)
        at androidx.appcompat.app.AppCompatDelegateImpl.setContentView(AppCompatDelegateImpl.java:470)
        at androidx.appcompat.app.AppCompatActivity.setContentView(AppCompatActivity.java:141)
        at com.otaliastudios.cameraview.demo.CameraActivity.onCreate(CameraActivity.java:57)
        at android.app.Activity.performCreate(Activity.java:7224)
        at android.app.Activity.performCreate(Activity.java:7213)
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1272)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2926)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3081)
        at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:78)
        at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:108)
[... 1159 lines cut ...]
2019-07-22 12:13:01.910 32339-32365/com.otaliastudios.cameraview.demo I/Step: BIND doStop Called. Enqueuing.
2019-07-22 12:13:01.910 32339-32365/com.otaliastudios.cameraview.demo I/Step: BIND doStop About to stop. Setting state to STOPPING
2019-07-22 12:13:01.910 32339-32365/com.otaliastudios.cameraview.demo I/Camera2Engine: onStopBind: About to clean up.
2019-07-22 12:13:01.911 32339-32365/com.otaliastudios.cameraview.demo I/Camera2Engine: onStopBind: Returning.
2019-07-22 12:13:01.911 32339-32365/com.otaliastudios.cameraview.demo I/Step: BIND doStop Succeeded! Setting state to STOPPED
2019-07-22 12:13:01.912 32339-32365/com.otaliastudios.cameraview.demo I/Step: ENGINE doStop Called. Enqueuing.
2019-07-22 12:13:01.912 32339-32365/com.otaliastudios.cameraview.demo I/Step: ENGINE doStop About to stop. Setting state to STOPPING
2019-07-22 12:13:01.912 32339-32365/com.otaliastudios.cameraview.demo I/Camera2Engine: onStopEngine: About to clean up.
2019-07-22 12:13:01.912 32339-32365/com.otaliastudios.cameraview.demo I/Camera2Engine: onStopEngine: Clean up. Releasing camera.
2019-07-22 12:13:01.921 32339-32339/com.otaliastudios.cameraview.demo W/ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@d5b897d
2019-07-22 12:13:02.118 32339-32357/com.otaliastudios.cameraview.demo E/libc: Access denied finding property "vendor.camera.aux.packagelist"
2019-07-22 12:13:02.112 32339-32339/com.otaliastudios.cameraview.demo W/Binder:32339_3: type=1400 audit(0.0:232568): avc: denied { read } for name="u:object_r:vendor_camera_prop:s0" dev="tmpfs" ino=13672 scontext=u:r:untrusted_app:s0:c247,c256,c512,c768 tcontext=u:object_r:vendor_camera_prop:s0 tclass=file permissive=0
2019-07-22 12:13:02.132 32339-32339/com.otaliastudios.cameraview.demo W/Binder:32339_3: type=1400 audit(0.0:232569): avc: denied { read } for name="u:object_r:vendor_camera_prop:s0" dev="tmpfs" ino=13672 scontext=u:r:untrusted_app:s0:c247,c256,c512,c768 tcontext=u:object_r:vendor_camera_prop:s0 tclass=file permissive=0
2019-07-22 12:13:02.141 32339-32357/com.otaliastudios.cameraview.demo E/libc: Access denied finding property "vendor.camera.aux.packagelist"
[... cut ...]
2019-07-22 12:13:02.267 32339-32339/com.otaliastudios.cameraview.demo D/MediaPlayerNative: getMetadata
2019-07-22 12:13:02.271 32339-32339/com.otaliastudios.cameraview.demo E/VideoPreview: The video full size is 720.0x960.0
2019-07-22 12:13:02.379 32339-32357/com.otaliastudios.cameraview.demo W/MediaPlayerNative: info/warning (3, 0)
2019-07-22 12:13:02.655 32339-32339/com.otaliastudios.cameraview.demo I/CameraEngine: onSurfaceDestroyed
2019-07-22 12:13:02.656 32339-32365/com.otaliastudios.cameraview.demo I/CameraEngine: stopPreview needsStopPreview: false swallowExceptions: false
```

natario1, 2019-07-22:
Thanks! There's not much I can do since I can't reproduce, I hope you can take a look into it.

I have no idea what this is about. Looks like a pixel format mismatch? I would be curious to know what colors you see if you draw some Color.RED, Color.GREEN and Color.BLUE in the overlay.

BlindJerobine, 2019-07-22:
I have the exact same problem with my Pixel 2XL, if I record a Video Snapshot.
But it works in my real App (Except for #518).

I think this could be Demo app exclusive.

natario1, 2019-07-22:
I would also be curious to know if it happens with Camera1 or Camera2 or both.

RAN3000, 2019-07-23:
I just checked, it happens with both Camera1 and Camera2.

I added an overlay with three squares with colors (from left to right): Color.RED, Color.GREEN, Color.BLUE.
Here's what I get:
[image]

natario1, 2019-07-23:
I see. I really don't know how to address this issue though.

natario1, 2019-07-24:
Do picture snapshots work well?

RAN3000, 2019-07-25:
They do work.

natario1, 2019-07-25:
@RAN3000 and does it works if you remove overlays?

sceneren, 2019-07-30:
I have the exact same problem with my Pixel XL, if I record a Video Snapshot.
but I remove overlays ,it can works

natario1, 2019-07-30:
I would appreciate if you guys could test the workaround-514 branch.

sceneren, 2019-07-31:
i tested the workaround-514 branch and saw the interface when I recorded the video, but sometimes it flickered or only recorded the first frame.

natario1, 2019-07-31:
@sceneren could you post a video of your screen with workaround-514?

sceneren, 2019-07-31:
[device-2019-07-31-154027.mp4.zip](https://github.com/natario1/CameraView/files/3450399/device-2019-07-31-154027.mp4.zip)

natario1, 2019-07-31:
Hey @sceneren could you pull again that branch and try again? I'm hoping it should be fixed now.

sceneren, 2019-07-31:
Thanks a lot, it works fine now in my pixel xl, and when can you merge it into the master branch

natario1, 2019-07-31:
In the next few days! Thanks for helping.

closed 2019-08-01 by natario1

RAN3000, 2019-08-01:
It works on my Xiaomi Redmi 7 too.

PR #528 (2019-08-01): Fix issue 514
- Fixes #514
- Improves picture snapshot recorder
- Improves thread management