natario1/CameraView #522: Both setPictureSize and setPreviewStreamSize have no effect on addFrameProcessor width and height

bilsou, 2019-07-26:
Using the code below on a Nokia 5.1 with Android Pie and CameraEngine2

```
        val width = SizeSelectors.minWidth(700)
        val height = SizeSelectors.minHeight(700)
        val dimensions = SizeSelectors.and(width, height)

        viewFinder.setPictureSize(dimensions)
        viewFinder.setPreviewStreamSize(dimensions)

       viewFinder.open()
        viewFinder.addFrameProcessor { frame ->
            val buffer = frame.byteBuffer
            val width = frame.size.width
            val height = frame.size.height
        }
```

Width and height are always 640*480 regardless of what is chosen.

natario1, 2019-07-27:
Yeah, with Camera2 you have no control over the frame processor size. There has never been an explicit API for that.

bilsou, 2019-07-27:
I remembered afterwards how unreliable Camera2 API was depending on the hardware/manufacturer compared to Camera1. I will stick with Camera1.

Thank you anyway for getting back to me.

closed 2019-07-27 by bilsou

natario1, 2019-07-27:
But we might add this in the future. I just think we'll keep a hard limit of 700 for both dimensions.