natario1/CameraView #525: Strange performance issue between 1.6.1 and 2.0.0-rc2

wosiu, 2019-07-28:
### Describe the bug
`takePictureSnapshot()` is ~2 time faster/slower between 1.6.1 and 2.0.0-rc2 depending on a device.

- CameraView version: 1.6.1 and 2.0.0-rc2
- Reproducible in official demo app: neet to investigate
- Device / Android version: many - check below

### To Reproduce
The way I'm using CameraView is:
1. take snapshot
2. process snapshot
3. go to 1.

One such iteration takes:

dev \ version | 1.6.1 | 2.0.0-rc2 
---- | ---- | ---- 
Huawei P10-lite (WAS-LX1), Android 7.0, API 24 | **~1.3 sec** | ~ 3.2 sec 
Sony Xperia M5, Android 6.0, API 23 | ~3.1 sec | **~1.2 sec** 
Xiaomi Redmi 6, Android 5.1.1, API 22 | ~2.5 sec | **~1.4 sec**

I repeated this test many times, in different orders and so on. So it is super strange that depending a device (or API?) there is a performance degradation or speed-up.

### Expected behavior
At least there should be no performance degradation after upgrading to 2.0.0-rc2.

I can help with investigating, but some hints where to start are welcomed.

natario1, 2019-07-28:
My first suggestion is to leave your processing out and do this in the demo app.

This said, snapshots are taken in a completely different way after v1.

wosiu, 2019-07-28:
I tested in the demo app (without any modifications), overlay in preview -> false. Given times in ms, average of several tries. For 1.6.1 only "picture" mode, as "snapshot" is not available in the demo app.

| device                                         | 1.6.1,  picture | 2.0.0-rc2,  snapshot,  camera1 | 2.0.0-rc2,  snapshot,  camera2 | 2.0.0-rc2,  picture,  camera1 | 2.0.0-rc2,  picture,  camera2 |
|------------------------------------------------|-----------------|--------------------------------|--------------------------------|-------------------------------|-------------------------------|
| Huawei P10-lite (WAS-LX1), Android 7.0, API 24 | 567             | **1477**                           | **1349**                           | 500                          | 1789                          |
| Sony Xperia M5, Android 6.0, API 23            | 909             | 335                            | 335                            | 1090                          | 1476                          |

2 conclusions:
1) engine `camera1` is faster for taking photos (v1.6.1 ~= v2.0.0-rc2)
2) there is indeed something wrong with taking snapshots for some devices - it takes way more time than taking photos regardless of the choice of an engine. I also tested against current master (3e0ae65) and it does not fix described behaviour.

natario1, 2019-07-28:
We do not even try to make pictures fast, but picture snapshots should be, faster than 1400 for sure. You could try

- removing snapshot overlays, this saves time 
- setting a snapshotMaxWidth or snapshotMaxHeight, this will help a lot too

And you can add logs for the device that has this issue and see where this time is being spent. I suspect that's EglBaseSurface.saveFrame, but it might be somewhere else. Anyway I fear this can not be optimized further.

wosiu, 2019-07-28:
Profiler (for current master, camera1/camera2, Huawei P10-lite WAS-LX1, Android 7.0, API 24) says it spends most of that time in "FallbackCameraT" thread during capturing a snapshot:

[image]

"FallbackCameraT" seems to be recreated for each snapshot call. Here I called 3 times "capture snapshot", one just after the others:

[image]

wosiu, 2019-07-28:
For Sony Xperia M5, Android 6.0, API 23 behaviour is similar, however time spent in "FallbackCameraT" is way shorter:
[image]

wosiu, 2019-07-28:
I think I found a wrongdoer:
[image]

EglBaseSurface -> saveFrameTo -> saveFrameTo()

natario1, 2019-07-29:
Yes, in your case I guess it is doing some color conversion on the CPU instead of GPU before returning the results.

wosiu, 2019-07-29:
> setting a snapshotMaxWidth or snapshotMaxHeight, this will help a lot too
that indeed helped a lot and probably I'll be fine.
However regarding:
> it is doing some color conversion on the CPU instead of GPU
Is there a chance that converting somehow to greyscale would be even faster? Because at the end of the day I need a greyscale image.

natario1, 2019-07-29:
In theory yes, but you would have to inject stuff at various parts of the library, use a GL shader to convert frames to grayscale, ensure surface format is grayscale (not sure if possible), and hope that glReadPixels supports reading that single channel, which I'm not sure, and hope that you can create a `Bitmap` out of that...

wosiu, 2019-07-29:
Got it. For now I'll stick with `snapshotMaxWidth` as further speed up is not that critical though - I went down to less than 200ms with snapshot width ~340px which is fine for my purpose.
I'm closing this one. Many thanks and great job btw :)

closed 2019-07-29 by wosiu