natario1/CameraView #526: camera.takeVideoSnapshot() crash

sceneren, 2019-07-30:
### Describe the bug
use camera.takeVideoSnapshot() after,click HOME to background  and restart APP ,crashed

- CameraView version: 2.0.0-rc2
- Reproducible in official demo app: yes
- Device / Android version: Google Pixel XL Android 9, API 28

###Log:
2019-07-30 17:03:57.582 3189-3265/com.agz.testvideorecord E/GLConsumer: [SurfaceTexture-2-3189-1] checkAndUpdateEglState: invalid current EGLContext
2019-07-30 17:03:57.602 3189-3265/com.agz.testvideorecord E/AndroidRuntime: FATAL EXCEPTION: GLThread 2856
    Process: com.agz.testvideorecord, PID: 3189
    java.lang.IllegalStateException: Unable to update texture contents (see logcat for details)
        at android.graphics.SurfaceTexture.nativeUpdateTexImage(Native Method)
        at android.graphics.SurfaceTexture.updateTexImage(SurfaceTexture.java:243)
        at com.otaliastudios.cameraview.video.SnapshotVideoRecorder.onRendererFrame(SnapshotVideoRecorder.java:181)
        at com.otaliastudios.cameraview.preview.GlCameraPreview$Renderer.onDrawFrame(GlCameraPreview.java:212)
        at android.opengl.GLSurfaceView$GLThread.guardedRun(GLSurfaceView.java:1571)
        at android.opengl.GLSurfaceView$GLThread.run(GLSurfaceView.java:1270)
2019-07-30 17:03:57.848 3189-3189/com.agz.testvideorecord E/libEGL: call to OpenGL ES API with no current context (logged once per thread)

natario1, 2019-07-30:
You press home while recording the video? Or what?

sceneren, 2019-07-30:
yes,when I recoding video ,click Home

closed 2019-08-01 by natario1

sceneren, 2019-08-01:
When to update Gradle dependencies？Fix #526,#514

natario1, 2019-08-01:
@sceneren you can Watch this repository so you'll get notifications when we have new releases.

sceneren, 2019-08-01:
@natario1 thx

commit 13387dc8 (2019-08-01): Fix #526
commit c824c7db (2019-08-01): Improve video snapshots (#530)

* Fix button elevation

* Replace zero buffer with sample noise

* Small improvements

* Drop frames when we have too many pending events

* Warmup worker threads, enlarge audio buffers, use EncoderEngine thread

* Improve audio noise

* Ensure first frame is a sync one

* Request key frames when the first is lost

* Fix README

* Create AudioNoise

* Correctly recycle OutputBuffer

* Fix #526