natario1/CameraView #532: How to make the captured photo or recorded video from the front camera looks like the preview not flipped?

Ma7moudHatem, 2019-08-05:
### How to make the captured photo or recorded video from the front camera looks like the preview not flipped?

When using the front camera and want to record a video or capturing a photo, the recorded video output or the captured photo output becomes flipped horizontally and not looks like what it was while recording or previewing.
So, can you please help me with my issue.

### Version used
CameraView 2.0.0

natario1, 2019-08-05:
You can use the snapshot APIs to catch exactly what you see.

Ma7moudHatem, 2019-08-05:
@natario1, what if I don't want to use the snapshot method because of the limitations in it, what can I do?

natario1, 2019-08-05:
You can post-process the JPEG picture or MP4 video to flip them, but you will have to use separate library/tools to do it

closed 2019-08-19 by natario1