natario1/CameraView #533: Getting frame size bigger than preview size in some devices

hmittal1990, 2019-08-05:
### Describe the bug
A clear and concise description of what the bug is.

- CameraView version: *version number*
- Reproducible in official demo app: *yes/no*
- Device / Android version: *Pixel, API 28*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Go to '...'
2. Click on '....'
3. See error

### Expected behavior
A clear and concise description of what you expected to happen.

### Screenshots
If applicable, add screenshots to help explain your problem.

### Logs
Use `CameraLogger.setLogLevel(LEVEL_VERBOSE)` to see all logs into LogCat.
Use `CameraLogger.registerLogger()` to export to file or crash reporting service.

### APK
Link to a Github repo where the bug is reproducible.

natario1, 2019-08-05:
Can you fill all the bug info? Thanks!

hmittal1990, 2019-08-05:
Info filled

natario1, 2019-08-05:
What is preview size? How are you getting it?

hmittal1990, 2019-08-05:
Preview size is the cameraview dimensions i am assuming.

natario1, 2019-08-05:
There can be a difference between the two, but the aspect ratio is always the same (4:3 in your case). So you can just scale down or up to match the other size without distortion.

hmittal1990, 2019-08-05:
Can you provide me any example ?

hmittal1990, 2019-08-05:
Thanks for the information

natario1, 2019-08-08:
Depends on what you are doing, I think you might find help asking on StackOverflow.

hmittal1990, 2019-08-08:
I have solved it with mathematical calculation. Thanks for the great library

closed 2019-08-08 by natario1