natario1/CameraView #534: Honor 10 Lite  width and height use match_parent  and set cameraAudio=true crashed

sceneren, 2019-08-06:
- CameraView version: 2.0.0
- Reproducible in official demo app: yes
- Device / Android version: Honor 10 Lite, API 28

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. set width and height  = match_parent
2. set cameraAudio=true
3.click snap video

### Logs
2019-08-06 11:44:36.355 28368-28405/com.otaliastudios.cameraview.demo E/HwCustAudioRecordImpl: isOpenEC : false
2019-08-06 11:44:36.393 28368-28581/com.otaliastudios.cameraview.demo E/HwExtendedCodec: mime: video/avc matching compontent failed!
2019-08-06 11:44:36.395 28368-28582/com.otaliastudios.cameraview.demo E/HwExtendedCodec: mime: audio/mp4a-latm matching compontent failed!
2019-08-06 11:44:36.414 28368-28585/com.otaliastudios.cameraview.demo E/ACodec: [OMX.hisi.video.encoder.avc] failed to set input port definition parameters.
2019-08-06 11:44:36.414 28368-28585/com.otaliastudios.cameraview.demo E/ACodec: [OMX.hisi.video.encoder.avc] configureCodec returning error -1010
2019-08-06 11:44:36.414 28368-28585/com.otaliastudios.cameraview.demo E/ACodec: signalError(omxError 0x80001001, internalError -1010)
2019-08-06 11:44:36.414 28368-28583/com.otaliastudios.cameraview.demo E/MediaCodec: Codec reported err 0xfffffc0e, actionCode 0, while in state 3
2019-08-06 11:44:36.418 28368-28581/com.otaliastudios.cameraview.demo E/MediaCodec: configure failed with err 0xfffffc0e, resetting...
2019-08-06 11:44:36.432 28368-28581/com.otaliastudios.cameraview.demo E/AndroidRuntime: FATAL EXCEPTION: VideoEncoder
    Process: com.otaliastudios.cameraview.demo, PID: 28368
    android.media.MediaCodec$CodecException: Error 0xfffffc0e
        at android.media.MediaCodec.native_configure(Native Method)
        at android.media.MediaCodec.configure(MediaCodec.java:1967)
        at android.media.MediaCodec.configure(MediaCodec.java:1896)
        at com.otaliastudios.cameraview.video.encoding.VideoMediaEncoder.onPrepare(VideoMediaEncoder.java:74)
        at com.otaliastudios.cameraview.video.encoding.TextureMediaEncoder.onPrepare(TextureMediaEncoder.java:97)
        at com.otaliastudios.cameraview.video.encoding.MediaEncoder$1.run(MediaEncoder.java:190)
        at android.os.Handler.handleCallback(Handler.java:891)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:207)
        at android.os.HandlerThread.run(HandlerThread.java:65)

natario1, 2019-08-08:
Can you identify whether your issue is video or audio or both?

- Does it work with audio=off / width and height=match_parent?
- Does it work with audio=on / width and height=wrap_content?
- Does it work with audio=off / width and height=wrap_content?

sceneren, 2019-08-08:
audio=on, width = match_parent, height= match_parent  don't work
audio=off, width = match_parent ,height= match_parent can work
audio=on, width = wrap_content, height= wrap_content  can work
audio=off, width = wrap_content, height= wrap_content can work

natario1, 2019-08-08:
Are you sure? The stack trace you posted is about video so I don't understand why `audio=on/off` should make a difference.

Just asking. Maybe you have different stack traces.

sceneren, 2019-08-09:
[SVID_20190809_110843_1.mp4.zip](https://github.com/natario1/CameraView/files/3484589/SVID_20190809_110843_1.mp4.zip)

u can see this video ,set width and heigth = match_parent

natario1, 2019-08-09:
Could you post all logs from the demo app from start to crash?

natario1, 2019-08-12:
@sceneren Could you also test the `video-encoders` branch?

sceneren, 2019-08-13:
I test this branch.
#####################
width and height =wrap_content
#####################

2019-08-13 14:16:58.842 5661-5661/com.otaliastudios.cameraview.demo D/HwAppInnerBoostImpl: asyncReportData com.otaliastudios.cameraview.demo,2,1,1,0 interval=67
2019-08-13 14:16:58.859 5661-5661/com.otaliastudios.cameraview.demo V/AudioManager: playSoundEffect   effectType: 0
2019-08-13 14:16:58.859 5661-5661/com.otaliastudios.cameraview.demo V/AudioManager: querySoundEffectsEnabled...
2019-08-13 14:16:58.861 5661-5661/com.otaliastudios.cameraview.demo W/DemoApp: Recording snapshot for 5 seconds...
2019-08-13 14:16:58.888 5661-5661/com.otaliastudios.cameraview.demo V/CameraEngine: takeVideoSnapshot scheduling
2019-08-13 14:16:58.889 5661-5704/com.otaliastudios.cameraview.demo V/CameraEngine: takeVideoSnapshot performing. BindState: 2 isTakingVideo: false
2019-08-13 14:16:58.890 5661-5704/com.otaliastudios.cameraview.demo I/Camera2Engine: onTakeVideoSnapshot rotation: 0 size: 1080x1440
2019-08-13 14:16:58.898 5661-5707/com.otaliastudios.cameraview.demo D/OpenGLRenderer:   HWUI Binary is  enabled
2019-08-13 14:16:58.898 5661-5707/com.otaliastudios.cameraview.demo D/OpenGLRenderer: disableOutlineDraw is true
2019-08-13 14:16:58.898 5661-5717/com.otaliastudios.cameraview.demo I/SnapshotVideoRecorder: Starting the encoder engine.
2019-08-13 14:16:58.908 5661-5707/com.otaliastudios.cameraview.demo D/OpenGLRenderer:   HWUI Binary is  enabled
2019-08-13 14:16:58.908 5661-5707/com.otaliastudios.cameraview.demo D/OpenGLRenderer: disableOutlineDraw is true
2019-08-13 14:16:58.934 5661-5707/com.otaliastudios.cameraview.demo D/mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, EGLBoolean) returns 0x3000
2019-08-13 14:16:58.941 5661-5717/com.otaliastudios.cameraview.demo W/VideoCapabilities: Unrecognized profile 4 for video/hevc
2019-08-13 14:16:58.943 5661-5717/com.otaliastudios.cameraview.demo W/VideoCapabilities: Unrecognized profile/level 0/0 for video/mpeg2
2019-08-13 14:16:58.943 5661-5717/com.otaliastudios.cameraview.demo W/VideoCapabilities: Unrecognized profile/level 0/2 for video/mpeg2
2019-08-13 14:16:58.943 5661-5717/com.otaliastudios.cameraview.demo W/VideoCapabilities: Unrecognized profile/level 0/3 for video/mpeg2
2019-08-13 14:16:58.943 5661-5717/com.otaliastudios.cameraview.demo I/VideoCapabilities: Unsupported profile 5 for video/mpeg2
2019-08-13 14:16:58.943 5661-5717/com.otaliastudios.cameraview.demo I/chatty: uid=10141(com.otaliastudios.cameraview.demo) GLThread 5259 identical 2 lines
2019-08-13 14:16:58.943 5661-5717/com.otaliastudios.cameraview.demo I/VideoCapabilities: Unsupported profile 5 for video/mpeg2
2019-08-13 14:16:58.945 5661-5717/com.otaliastudios.cameraview.demo W/VideoCapabilities: Unrecognized profile/level 1/32 for video/mp4v-es
2019-08-13 14:16:58.945 5661-5717/com.otaliastudios.cameraview.demo W/VideoCapabilities: Unrecognized profile/level 32768/2 for video/mp4v-es
2019-08-13 14:16:58.945 5661-5717/com.otaliastudios.cameraview.demo W/VideoCapabilities: Unrecognized profile/level 32768/64 for video/mp4v-es
2019-08-13 14:16:58.965 5661-5717/com.otaliastudios.cameraview.demo I/VideoCapabilities: Unsupported profile 4 for video/mp4v-es
2019-08-13 14:16:58.971 5661-5717/com.otaliastudios.cameraview.demo I/DeviceEncoders: findDeviceEncoder - type: video/avc encoders: 2
2019-08-13 14:16:58.972 5661-5717/com.otaliastudios.cameraview.demo I/DeviceEncoders: Enabled. Found video encoder: OMX.hisi.video.encoder.avc
2019-08-13 14:16:58.972 5661-5717/com.otaliastudios.cameraview.demo I/DeviceEncoders: findDeviceEncoder - type: audio/mp4a-latm encoders: 1
2019-08-13 14:16:58.973 5661-5717/com.otaliastudios.cameraview.demo I/DeviceEncoders: Enabled. Found audio encoder: OMX.google.aac.encoder
2019-08-13 14:16:58.977 5661-5717/com.otaliastudios.cameraview.demo W/libEGL: EGLNativeWindowType 0x7dd92f5010 disconnect failed
    
    --------- beginning of crash
2019-08-13 14:16:58.979 5661-5717/com.otaliastudios.cameraview.demo E/AndroidRuntime: FATAL EXCEPTION: GLThread 5259
    Process: com.otaliastudios.cameraview.demo, PID: 5661
    java.lang.RuntimeException: Height not supported after adjustment. Desired:1440 Range:[144, 1088]
        at com.otaliastudios.cameraview.internal.DeviceEncoders.getSupportedVideoSize(DeviceEncoders.java:151)
        at com.otaliastudios.cameraview.video.SnapshotVideoRecorder.onRendererFrame(SnapshotVideoRecorder.java:136)
        at com.otaliastudios.cameraview.preview.GlCameraPreview$Renderer.onDrawFrame(GlCameraPreview.java:217)
        at android.opengl.GLSurfaceView$GLThread.guardedRun(GLSurfaceView.java:1609)
        at android.opengl.GLSurfaceView$GLThread.run(GLSurfaceView.java:1293)
2019-08-13 14:16:58.988 5661-5661/com.otaliastudios.cameraview.demo I/CameraEngine: Stop: posting runnable. State: ENGINE_STATE_STARTED
2019-08-13 14:16:58.989 5661-5704/com.otaliastudios.cameraview.demo W/CameraEngine: Stop: executing runnable. AllState is 2
2019-08-13 14:16:58.990 5661-5661/com.otaliastudios.cameraview.demo D/ZrHung.AppEyeUiProbe: stop checker.
2019-08-13 14:16:58.990 5661-5704/com.otaliastudios.cameraview.demo I/Step: ALL doStop Called. Enqueuing.
2019-08-13 14:16:58.990 5661-5704/com.otaliastudios.cameraview.demo I/Step: ALL doStop About to stop. Setting state to STOPPING
[... cut ...]
2019-08-13 14:19:37.385 6462-6491/com.otaliastudios.cameraview.demo I/BackgroundPermManager: pkgName: com.otaliastudios.cameraview.demo,pid: 6462 ,uidOf3RdApk: 10141 ,permType: 0 ,permCfg: 0
2019-08-13 14:19:37.388 6462-6491/com.otaliastudios.cameraview.demo I/HwCameraUtil: notifySurfaceFlingerCameraStatus : isFront = false , isOpend = false
2019-08-13 14:19:37.393 6462-6471/com.otaliastudios.cameraview.demo W/System: A resource failed to call release. 
2019-08-13 14:19:37.456 6462-6517/com.otaliastudios.cameraview.demo I/Process: Sending signal. PID: 6462 SIG: 9

natario1, 2019-08-13:
@sceneren I have updated the branch. Can you try again? I think this should be fixed now.

sceneren, 2019-08-19:
I'm sorry to have taken so long to reply to you. Now it can work properly.

natario1, 2019-08-19:
Thanks! I'll close this when we merge the PR

PR #545 (2019-08-21): Encoders improvements
- Checks encoders using `MediaCodecList`, preferring hardware accelerated encoders over the software ones
- Aligns width, height and bitRate based on codec requirements.

Fixes #534
Fixes #548
closed 2019-08-21 by natario1