natario1/CameraView #539: Getting FATAL exception while coming back to fragment after captured video

narayan07feb, 2019-08-08:
In my app there is two fragment, In one fragment i am using cameraView to capture video , after video captured, i am replacing fragment to another fragment which playing recorded file,  when i am pressing back button i am getting below exception

Process: com.samyotech.keymarket, PID: 26451
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Internal Camera1 error. 2
        at com.otaliastudios.cameraview.engine.Camera1Engine.onError(Camera1Engine.java:92)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1342)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)
     Caused by: java.lang.RuntimeException: Internal Camera1 error. 2
        at com.otaliastudios.cameraview.engine.Camera1Engine.onError(Camera1Engine.java:85)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1342) 
        at android.os.Handler.dispatchMessage(Handler.java:106) 
        at android.os.Looper.loop(Looper.java:193) 
        at android.os.HandlerThread.run(HandlerThread.java:65) 
2019-08-08 09:25:07.398 26451-27694/com.samyotech.keymarket E/CameraEngine: uncaughtException: Got CameraException: com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Internal Camera1 error. 2 on engine state: ENGINE_STATE_STARTED
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Internal Camera1 error. 2
        at com.otaliastudios.cameraview.engine.Camera1Engine.onError(Camera1Engine.java:92)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1342)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)
     Caused by: java.lang.RuntimeException: Internal Camera1 error. 2
        at com.otaliastudios.cameraview.engine.Camera1Engine.onError(Camera1Engine.java:85)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1342) 
        at android.os.Handler.dispatchMessage(Handler.java:106) 
        at android.os.Looper.loop(Looper.java:193) 
        at android.os.HandlerThread.run(HandlerThread.java:65) 

- CameraView version: 2.0.0-rc2

- Device / Android version: *Mi A1, API 28*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Go to '...'
2. Click on '....'
3. See error

natario1, 2019-08-08:
Most likely you are not releasing the CameraView correctly in the first fragment. Make sure you are using `setLifecycleOwner(getViewLifecycleOwner())` instead of `setLifecycleOwner(this)`.

narayan07feb, 2019-08-08:
tried same thing but still getting same error 
 override fun onViewCreated(view: View, savedInstanceState: Bundle?) {
        super.onViewCreated(view, savedInstanceState)
        preview.setLifecycleOwner(viewLifecycleOwner);
        img_switch_camera.setOnClickListener(::switchCamera);
        img_flash_control.setOnClickListener(::controlflash);
        center_progress_button.setRecordListener(recordingListiner);
        preview.addCameraListener(cameraListener);
    }

closed 2019-08-08 by narayan07feb