natario1/CameraView #542: Layout Editor freezes "Waiting for build to finish"

iRYO400, 2019-08-09:
### Describe the bug
After opening a xml layout that contains CameraView, Android Studio's layout editor change state to "Waiting for build to finish" for all projects on my mac. The only thing can help is Invalidate cache and restart

- CameraView version: 2.1.0/2.0.0
- Reproducible in official demo app: *yes*
- Android Studio version: 3.4.2(stable)/3.5 RC2
- macOS Mojave 10.14.5/10.14.6

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Open activity_camera.xml
2. See text "Waiting for build to finish"

[image]

iRYO400, 2019-08-09:
Just tested on Windows 10 PC at work. Same result
Android Studio 3.4.2(stable)

iRYO400, 2019-08-10:
Just caught this in Layout Editor, dunno is it related to view
```
Write access not allowed during rendering (/Users/akbolat_ss/Library/Caches/AndroidStudio3.5/caches/corruption.marker)
	at com.android.tools.idea.rendering.RenderSecurityException.create(RenderSecurityException.java:52)
	at com.android.tools.idea.rendering.RenderSecurityManager.checkWrite(RenderSecurityManager.java:564)
	at java.io.FileOutputStream.<init>(FileOutputStream.java:200)
	at java.io.FileOutputStream.<init>(FileOutputStream.java:162)
	at java.io.FileWriter.<init>(FileWriter.java:90)
	at com.intellij.openapi.vfs.newvfs.persistent.FSRecords$DbConnection.createBrokenMarkerFile(FSRecords.java:213)
	at com.intellij.openapi.vfs.newvfs.persistent.FSRecords$DbConnection.handleError(FSRecords.java:522)
	at com.intellij.openapi.vfs.newvfs.persistent.FSRecords$DbConnection.access$000(FSRecords.java:154)
	at com.intellij.openapi.vfs.newvfs.persistent.FSRecords.writeAndHandleErrors(FSRecords.java:981)
	at com.intellij.openapi.vfs.newvfs.persistent.FSRecords.access$300(FSRecords.java:48)
	at com.intellij.openapi.vfs.newvfs.persistent.FSRecords$ContentOutputStream.writeBytes(FSRecords.java:1541)
	at com.intellij.openapi.vfs.newvfs.persistent.FSRecords$ContentOutputStream.access$3200(FSRecords.java:1522)
	at com.intellij.openapi.vfs.newvfs.persistent.FSRecords.writeContent(FSRecords.java:1487)
	at com.intellij.openapi.vfs.newvfs.persistent.PersistentFSImpl.writeContent(PersistentFSImpl.java:246)
	at com.intellij.openapi.vfs.newvfs.persistent.PersistentFSImpl.contentsToByteArray(PersistentFSImpl.java:524)
	at com.intellij.openapi.vfs.newvfs.impl.VirtualFileImpl.contentsToByteArray(VirtualFileImpl.java:139)
	at com.intellij.openapi.vfs.newvfs.impl.VirtualFileImpl.contentsToByteArray(VirtualFileImpl.java:130)
	at com.android.tools.idea.rendering.RenderClassLoader.loadClassFile(RenderClassLoader.java:148)
	at org.jetbrains.android.uipreview.ModuleClassLoader.loadClassFile(ModuleClassLoader.java:312)
	at org.jetbrains.android.uipreview.ModuleClassLoader.loadClassFromModuleOrDependency(ModuleClassLoader.java:250)
	at org.jetbrains.android.uipreview.ModuleClassLoader.load(ModuleClassLoader.java:188)
	at com.android.tools.idea.rendering.RenderClassLoader.findClass(RenderClassLoader.java:61)
	at org.jetbrains.android.uipreview.ModuleClassLoader.findClass(ModuleClassLoader.java:135)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at org.jetbrains.android.uipreview.ModuleClassLoader.loadClass(ModuleClassLoader.java:235)
	at androidx.constraintlayout.widget.ConstraintLayout.init(ConstraintLayout.java:599)
	at androidx.constraintlayout.widget.ConstraintLayout.<init>(ConstraintLayout.java:576)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at org.jetbrains.android.uipreview.ViewLoader.createNewInstance(ViewLoader.java:403)
	at org.jetbrains.android.uipreview.ViewLoader.loadClass(ViewLoader.java:186)
	at org.jetbrains.android.uipreview.ViewLoader.loadView(ViewLoader.java:144)
	at com.android.tools.idea.rendering.LayoutlibCallbackImpl.loadView(LayoutlibCallbackImpl.java:309)
	at android.view.BridgeInflater.loadCustomView(BridgeInflater.java:418)
	at android.view.BridgeInflater.loadCustomView(BridgeInflater.java:429)
	at android.view.BridgeInflater.createViewFromTag(BridgeInflater.java:333)
	at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:730)
	at android.view.LayoutInflater.inflate(LayoutInflater.java:492)
	at android.view.LayoutInflater.inflate(LayoutInflater.java:394)
	at com.android.layoutlib.bridge.impl.RenderSessionImpl.inflate(RenderSessionImpl.java:323)
	at com.android.layoutlib.bridge.Bridge.createSession(Bridge.java:394)
	at com.android.tools.idea.layoutlib.LayoutLibrary.createSession(LayoutLibrary.java:200)
	at com.android.tools.idea.rendering.RenderTask.createRenderSession(RenderTask.java:572)
	at com.android.tools.idea.rendering.RenderTask.lambda$inflate$5(RenderTask.java:698)
	at java.util.concurrent.CompletableFuture$AsyncSupply.run(CompletableFuture.java:1590)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)

```

layout structure:
- layout
 - ConstraintLayout
  - CameraView
  - UI to control CameraView
 - /Constraint
- /layout

iRYO400, 2019-08-10:
Anybody, please help! 
Found that dont need to invalidate cache, "Just restart" can help

iRYO400, 2019-08-11:
Just commented lines 151 and 156, calling init method inside CameraView's constructors, now it doesn't freeze. It means init method is overloaded

closed 2019-08-12 by iRYO400

dssob, 2019-08-18:
Hello, I have faced the same issue, could you please give a piece of advice on how can I solve it?

iRYO400, 2019-08-19:
@dssob Hello! 
I removed CameraView from xml and initializing it programmatically
Example:
```
private lateinit var cameraView: CameraView
...
private fun initCamera() {
        cameraView = CameraView(context!!)
        cameraView.layoutParams = FrameLayout.LayoutParams(
            FrameLayout.LayoutParams.MATCH_PARENT,
            FrameLayout.LayoutParams.MATCH_PARENT
        )
        ... 
        // other CameraView settings, like mode, engine, preview, audio, etc.
        ...
        binding.root.addView(cameraView, 0) // adding CameraView to my layout, 0 means index. Set to 0, because I need it at the background of all views 
}
```

`initCamera` called in onCreate(in Activity) or onCreateView/onViewCreated(in Fragment)

ruidge, 2019-08-21:
@iRYO400 this xml preview issue has any effect on function while use api ?

iRYO400, 2019-08-21:
> @iRYO400 this xml preview issue has any effect on function while use api ?

 I think no. But if it would, just check `init` method and find which of initializers do not work

mosamman, 2019-10-27:
This issue shouldn't be marked as closed. Also @iRYO400 work around works.