natario1/CameraView #543: toggleFacing() while takings video

akokulyuk, 2019-08-09:
Hi, guys is it possible to switch facing without stopping recording?
Now if i toogleFacing() I just get video only from 1 facing, rest of the video is not available

### Version used
v. 2.1.0

natario1, 2019-08-10:
No, sorry!

JeevanDeep, 2019-08-10:
Hi, is it not possible to do even in snapshot mode?

natario1, 2019-08-10:
If you fork the project and change a few things you can make it work for video snapshots.

But we don't support this officially because changing the `Facing` value can change the view size, change the preview stream size and aspect ratio, and cause issues in the output video.

JeevanDeep, 2019-08-10:
@natario1 Can you guide me towards this approach? I forked the project and found that you restart the camera when switching the facing. Do i have to change something in stop and start function?

natario1, 2019-08-10:
@JeevanDeep it's not an easy task, I can't really write it in a comment... and I have never tried to do so. You should make sure the video recorder is not stopped, and the preview stream size stays consistent or you account for changes in it.

closed 2019-08-19 by natario1

denizsubasi, 2019-09-30:
is there any development about this feature ?

BhavyaRattan, 2020-06-26:
@natario1 do we support this behaviour now, I wanted to add same functionality in my app?