natario1/CameraView #544: Error while using

KanwarpartapSinghBimrah, 2019-08-12:
See screenshot what code I had written. I got only one error like this 

**java.lang.IllegalStateException: frame.size must not be null**
```
FATAL EXCEPTION: FrameProcessorsWorker
Process: com.royalbimrah.ai, PID: 12887
java.lang.IllegalStateException: frame.size must not be null
  at com.royalbimrah.ai.facedetection.FaceDetectionActivity.process(FaceDetectionActivity.kt:67)
  at com.otaliastudios.cameraview.CameraView$Callbacks$13.run(CameraView.java:1742)
  at android.os.Handler.handleCallback(Handler.java:751)
  at android.os.Handler.dispatchMessage(Handler.java:95)
  at android.os.Looper.loop(Looper.java:154)
  at android.os.HandlerThread.run(HandlerThread.java:61)
```

Also, you can see code and let me know how to solve this error soon.

[image]

natario1, 2019-08-25:
@KanwarpartapSinghBimrah are you using Camera1 or Camera2 engine?

Can you enable logs in your app (`CameraLogger.setLogLevel(VERBOSE)`) and paste them here?

natario1, 2019-08-25:
OK, I can see this issue with Camera2... I would like to know if you're using Camera1 or Camera2. If you are using Camera1, please post logs as well!

PR #572 (2019-08-29): Frame improvements
Improvements to frame processing should fix some crashes or at least give clear messages about what's happening.

Fixes #544
Fixes #554
commit eddae18c (2019-08-29): Frame improvements (#572)

* Fix #544

* Improve Frames behavior and error messages

* Improve Frames documentation

* Fix tests

* Fix video crashes
closed 2019-08-29 by natario1