natario1/CameraView #546: Crash when opening the camera more than once for <1% of users

vpmalley, 2019-08-13:
### Describe the bug
On Firebase Crashlytics, I notice a crash happening to <1% of users. This happens when the camera is opened twice without being closed

- CameraView version: *2.0.0-beta04*
- Reproducible in official demo app: *no* (Where can I find this demo app?)
- Device / Android version: *API 28, many devices, mostly Samsung but that might just be representative of the app users*

### To Reproduce
The CameraView is used in a fragment. It uses the fragment as its lifecycle owner. It uses a frame processor.
1. Open once a fragment with a CameraView in it.
2. The CameraView listener triggers twice `onCameraOpened` within 3-8 seconds (and no `onCameraClosed` nor `onCameraError`)
3. The app crashes with the logs below

This scenario is based on Crashlytics logs and playing with this screen manually, I've not reproduced it. It only happens to less than 1% of users, with the only significant criterion being the version of Android (9, API 28). I'd be happy to know when to catch this exception to recover properly.
The code makes a call to open the CameraView (`CameraView.open()`), which is now protected by `CameraView.isOpened` (and I've noticed fewer crashes, but not zero yet).

### Expected behavior
The CameraView shows the preview and does not crash.

### Screenshots
None available.

### Logs
Stacktrace on Crashlytics:
`Fatal Exception: java.lang.RuntimeException: getParameters failed (empty parameters)
       at android.hardware.Camera.native_getParameters(Camera.java)
       at android.hardware.Camera.getParameters + 2133(Camera.java:2133)
       at com.otaliastudios.cameraview.Camera1.startPreview + 186(Camera1.java:186)
       at com.otaliastudios.cameraview.Camera1.access$700 + 24(Camera1.java:24)
       at com.otaliastudios.cameraview.Camera1$3.run + 81(Camera1.java:81)
       at com.otaliastudios.cameraview.Camera1$2.run + 62(Camera1.java:62)
       at android.os.Handler.handleCallback + 873(Handler.java:873)
       at android.os.Handler.dispatchMessage + 99(Handler.java:99)
       at android.os.Looper.loop + 214(Looper.java:214)
       at android.os.HandlerThread.run + 65(HandlerThread.java:65)`

Crashlytics logs:
13:44:43 D/CrashlyticsCore Opened camera
13:44:46 D/CrashlyticsCore Opened camera

### APK
Public app: https://play.google.com/store/apps/details?id=com.flamingoscooters.android

natario1, 2019-08-13:
> It uses the fragment as its lifecycle owner

You should use `getViewLifecycleOwner()` and this might fix everything.

> The code makes a call to open the CameraView

When? From which thread? Also, if you use a lifecycle owner you don't need to open explicitly.

vpmalley, 2019-08-13:
> You should use `getViewLifecycleOwner()` and this might fix everything.
Thank you, this sounds like a good idea indeed. This looks like it's working with the first tests I made.
> When (do you open the CameraView)? From which thread? Also, if you use a lifecycle owner you don't need to open explicitly.
The scenario is that the CameraView can be hidden. When I hide it (setting the visibility to `GONE`) I also close it, and reopen when showing it again (all this is happening in the same fragment). 

Since that might be the cause of the crash, I might as well just not close it (I'm not sure how wise it is to keep it open while hidden, with a few tests it looks like performance-wise it's all fine). 

Thank you for all this advice, that sounds like it's likely to solve (or at least reduce) the amount of issues.

I just reproduced the crash (not consistently though) by quickly turning on the torch while the camera is open, then switching to the view where the CameraView is closed, then switching back to the open CameraView and that crashed at that point.
I don't know if that helps you locating some possible edge case when that can crash, otherwise feel free to close that issue, I would reopen it only if the crash happens while I never call `open()`

natario1, 2019-08-14:
Thanks for info @vpmalley ! I don't think this is worth investigating from our side, we expect `open()` to be called in a valid state and using the fragment as a lifecycle owner won't do this.

Feel free to `close()` and `open()` when changing visibility, as long as you change this while the fragment is attached (`fragment.getView() != null && fragment.getActivity() != null` might be a good check).

closed 2019-08-14 by natario1