natario1/CameraView #548: Unable to record video

vikas-markk, 2019-08-13:
###Getting error code -38, refer below logs.
E/MediaRecorder: start failed: -38
FullVideoRecorder: start: Error while starting media recorder. java.lang.IllegalStateException
java.lang.IllegalStateException
at android.media.MediaRecorder.start(Native Method)
at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:138)
at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:62)
at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:340)
at com.otaliastudios.cameraview.engine.CameraEngine$23.run(CameraEngine.java:1150)
at android.os.Handler.handleCallback(Handler.java:873)
at android.os.Handler.dispatchMessage(Handler.java:99)
at android.os.Looper.loop(Looper.java:193)
at android.os.HandlerThread.run(HandlerThread.java:65)
2019-08-13 11:42:24.234 2710-3054/I/CameraCallbacks: dispatchOnVideoRecordingEnd
2019-08-13 11:42:24.234 2710-3054/ E/MediaRecorder: stop called in an invalid state: 0

stop: Error while closing media recorder. java.lang.IllegalStateException
java.lang.IllegalStateException
at android.media.MediaRecorder.stop(Native Method)
at com.otaliastudios.cameraview.video.FullVideoRecorder.onStop(FullVideoRecorder.java:153)
at com.otaliastudios.cameraview.video.VideoRecorder.stop(VideoRecorder.java:69)
at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:144)
at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:62)
at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:340)
at com.otaliastudios.cameraview.engine.CameraEngine$23.run(CameraEngine.java:1150)
at android.os.Handler.handleCallback(Handler.java:873)
at android.os.Handler.dispatchMessage(Handler.java:99)
at android.os.Looper.loop(Looper.java:193)
at android.os.HandlerThread.run(HandlerThread.java:65)

CameraEngine: onVideoResult result is null: something went wrong. java.lang.IllegalStateException
java.lang.IllegalStateException
at android.media.MediaRecorder.start(Native Method)
at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:138)
at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:62)
at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:340)
at com.otaliastudios.cameraview.engine.CameraEngine$23.run(CameraEngine.java:1150)
at android.os.Handler.handleCallback(Handler.java:873)
at android.os.Handler.dispatchMessage(Handler.java:99)
at android.os.Looper.loop(Looper.java:193)
at android.os.HandlerThread.run(HandlerThread.java:65)

CameraView version: version number
Reproducible in official demo app: yes/no
Device / Android version: Pixel, API 28
To Reproduce
Steps to reproduce the behavior, possibly in the demo app:

Go to '...'
Click on '....'
See error
Expected behavior
A clear and concise description of what you expected to happen.

Screenshots
If applicable, add screenshots to help explain your problem.

Logs
Use CameraLogger.setLogLevel(LEVEL_VERBOSE) to see all logs into LogCat.
Use CameraLogger.registerLogger() to export to file or crash reporting service.

APK
Link to a Github repo where the bug is reproducible.

- CameraView version: *com.otaliastudios:cameraview:2.0.0-rc2*
- Reproducible in official demo app: *no*
- Device / Android version: *Pixel XL, API 28*

vikas-markk, 2019-08-13:
<com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        android:keepScreenOn="true"
        app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker"
        app:cameraFlash="auto"
        app:cameraGesturePinch="zoom"
        app:cameraGestureTap="autoFocus"
        app:cameraWhiteBalance="auto">

natario1, 2019-08-13:
Can you investigate as to why it's not reproducible in the demo app? 
There must be something in your app.

vikas-markk, 2019-08-13:
Just now added one fix in XML, then pixel its working fine .

    <com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        android:keepScreenOn="true"
        app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker"
        app:cameraFlash="auto"
        app:cameraEngine="camera2"
        app:cameraVideoBitRate="6400000"
        app:cameraGesturePinch="zoom"
        app:cameraGestureTap="autoFocus"
        app:cameraWhiteBalance="auto">

natario1, 2019-08-13:
@vikas-markk could you try with the [video-encoders](https://github.com/natario1/CameraView/tree/video-encoders) branch and see if it works without specifying the video bit rate?

vikas-markk, 2019-08-16:
@natario1 Yes its working on pixel, i wanted to test it on other device also.
Soon i will update you other device report also.

natario1, 2019-08-16:
Thank you @vikas-markk !

PR #545 (2019-08-21): Encoders improvements
- Checks encoders using `MediaCodecList`, preferring hardware accelerated encoders over the software ones
- Aligns width, height and bitRate based on codec requirements.

Fixes #534
Fixes #548
closed 2019-08-21 by natario1