natario1/CameraView #549: Video Recording crashes on HARDWARE_LEVEL_LEGACY devices

natario1, 2019-08-13:
### Describe the bug
When ending a normal video, the following exception is thrown:

```
2019-08-13 15:50:34.155 2835-9881/in.mohalla.sharechat E/AndroidRuntime: FATAL EXCEPTION: LegacyCameraCallback
    Process: in.mohalla.sharechat, PID: 2835
    java.lang.NullPointerException: Attempt to invoke virtual method 'android.hardware.camera2.CaptureRequest android.hardware.camera2.impl.CameraDeviceImpl$CaptureCallbackHolder.getRequest(int)' on a null object reference
        at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.onResultReceived(CameraDeviceImpl.java:1758)
        at android.hardware.camera2.legacy.CameraDeviceUserShim$CameraCallbackThread$CallbackHandler.handleMessage(CameraDeviceUserShim.java:316)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:154)
        at android.os.HandlerThread.run(HandlerThread.java:61) 
```

- CameraView version: 2.1.0
- Reproducible in official demo app: not sure
- Device / Android version: HARDWARE_LEVEL_LEGACY devices

### Logs

```
2019-08-13 15:50:30.358 2835-9930/in.mohalla.sharechat I/CameraDeviceState: Legacy camera service transitioning to state CAPTURING
2019-08-13 15:50:30.399 2835-9929/in.mohalla.sharechat W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
2019-08-13 15:50:30.399 2835-9929/in.mohalla.sharechat W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-08-13 15:50:30.399 2835-9929/in.mohalla.sharechat W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-08-13 15:50:30.856 2835-9877/in.mohalla.sharechat I/CameraCallbacks: dispatchOnVideoRecordingStart
2019-08-13 15:50:33.370 2835-2835/in.mohalla.sharechat D/AlarmPingSender: Sending Ping at:1565691633370
2019-08-13 15:50:33.371 2835-2835/in.mohalla.sharechat D/AlarmPingSender: Schedule next alarm at 1565691693371
2019-08-13 15:50:33.371 2835-2835/in.mohalla.sharechat D/AlarmPingSender: Alarm scheule using setExactAndAllowWhileIdle, next: 60000
2019-08-13 15:50:33.413 2835-3081/in.mohalla.sharechat D/AlarmPingSender: Success. Release lock(MqttService.client.user_70914340_143392be13733c298fd7):1565691633413
2019-08-13 15:50:33.585 2835-2835/in.mohalla.sharechat I/CameraEngine: stopVideo posting
2019-08-13 15:50:33.585 2835-9877/in.mohalla.sharechat I/RequestQueue: Repeating capture request cancelled.
2019-08-13 15:50:33.585 2835-9877/in.mohalla.sharechat I/RequestQueue: Repeating capture request set.
2019-08-13 15:50:33.586 2835-9877/in.mohalla.sharechat I/CameraEngine: stopVideo executing. isTakingVideo? true
2019-08-13 15:50:33.586 2835-9877/in.mohalla.sharechat I/CameraEngine: stopVideo executing. isTakingVideo? true
2019-08-13 15:50:33.586 2835-9877/in.mohalla.sharechat I/CameraCallbacks: dispatchOnVideoRecordingEnd
2019-08-13 15:50:33.604 2835-9929/in.mohalla.sharechat W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
2019-08-13 15:50:33.604 2835-9929/in.mohalla.sharechat W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-08-13 15:50:33.604 2835-9929/in.mohalla.sharechat W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-08-13 15:50:33.768 2835-9877/in.mohalla.sharechat I/RequestQueue: Repeating capture request cancelled.
2019-08-13 15:50:33.768 2835-9877/in.mohalla.sharechat I/RequestQueue: Repeating capture request set.
2019-08-13 15:50:33.768 2835-9877/in.mohalla.sharechat I/CameraCallbacks: dispatchOnVideoTaken com.otaliastudios.cameraview.VideoResult$Stub@2ae5b10
2019-08-13 15:50:33.781 2835-9930/in.mohalla.sharechat I/Adreno: DequeueBuffer: dequeueBuffer failed
2019-08-13 15:50:33.784 2835-9930/in.mohalla.sharechat W/SurfaceTextureRenderer: Surface abandoned, dropping frame. 
    android.hardware.camera2.legacy.LegacyExceptionUtils$BufferQueueAbandonedException
        at android.hardware.camera2.legacy.SurfaceTextureRenderer.swapBuffers(SurfaceTextureRenderer.java:533)
        at android.hardware.camera2.legacy.SurfaceTextureRenderer.drawIntoSurfaces(SurfaceTextureRenderer.java:751)
        at android.hardware.camera2.legacy.GLThreadManager$1.handleMessage(GLThreadManager.java:105)
        at android.os.Handler.dispatchMessage(Handler.java:98)
        at android.os.Looper.loop(Looper.java:154)
        at android.os.HandlerThread.run(HandlerThread.java:61)
2019-08-13 15:50:33.787 2835-9929/in.mohalla.sharechat E/RequestQueue: cancel failed: no repeating request exists for request id: 3
2019-08-13 15:50:33.792 2835-9929/in.mohalla.sharechat W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
2019-08-13 15:50:33.792 2835-9929/in.mohalla.sharechat W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-08-13 15:50:33.792 2835-9929/in.mohalla.sharechat W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-08-13 15:50:33.869 2835-10603/in.mohalla.sharechat D/OkHttp: --> POST https://tracking.sharechat.com/requestType16
2019-08-13 15:50:33.869 2835-10603/in.mohalla.sharechat D/OkHttp: Content-Type: application/json; charset=UTF-8
2019-08-13 15:50:33.869 2835-10603/in.mohalla.sharechat D/OkHttp: Content-Length: 310
2019-08-13 15:50:33.870 2835-10603/in.mohalla.sharechat D/OkHttp: {"bn":"broker1","client":"android","deviceId":"23fc4b3f3059191a","message":{"batchRequest":1,"requestArray":[{"language":"Hindi","time":1565691629725,"userId":"70914340","t":1565691629725,"i":136}]},"passCode":"143392be13733c298fd7","resTopic":"response/user_70914340_143392be13733c298fd7","userId":"70914340"}
2019-08-13 15:50:33.870 2835-10603/in.mohalla.sharechat D/OkHttp: --> END POST (310-byte body)
2019-08-13 15:50:33.982 2835-10603/in.mohalla.sharechat D/OkHttp: <-- 200 https://tracking.sharechat.com/requestType16 (111ms)
2019-08-13 15:50:33.982 2835-10603/in.mohalla.sharechat D/OkHttp: date: Tue, 13 Aug 2019 10:20:33 GMT
2019-08-13 15:50:33.982 2835-10603/in.mohalla.sharechat D/OkHttp: content-type: application/json;charset=utf-8
2019-08-13 15:50:33.982 2835-10603/in.mohalla.sharechat D/OkHttp: x-powered-by: Express
2019-08-13 15:50:33.982 2835-10603/in.mohalla.sharechat D/OkHttp: etag: W/"e-skwZlgCMCy1nW+FHck7oqs3GShc"
2019-08-13 15:50:33.982 2835-10603/in.mohalla.sharechat D/OkHttp: {"payload":{}}
2019-08-13 15:50:33.983 2835-10603/in.mohalla.sharechat D/OkHttp: <-- END HTTP (14-byte body)
2019-08-13 15:50:33.984 2835-10603/in.mohalla.sharechat W/FirebaseRemoteConfig: No value of type 'String' exists for parameter key 'sessions_max_length_minutes'.
2019-08-13 15:50:33.984 2835-9846/in.mohalla.sharechat W/FirebaseRemoteConfig: No value of type 'String' exists for parameter key 'sessions_feature_enabled'.
2019-08-13 15:50:34.155 2835-9881/in.mohalla.sharechat E/AndroidRuntime: FATAL EXCEPTION: LegacyCameraCallback
    Process: in.mohalla.sharechat, PID: 2835
    java.lang.NullPointerException: Attempt to invoke virtual method 'android.hardware.camera2.CaptureRequest android.hardware.camera2.impl.CameraDeviceImpl$CaptureCallbackHolder.getRequest(int)' on a null object reference
        at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.onResultReceived(CameraDeviceImpl.java:1758)
        at android.hardware.camera2.legacy.CameraDeviceUserShim$CameraCallbackThread$CallbackHandler.handleMessage(CameraDeviceUserShim.java:316)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:154)
        at android.os.HandlerThread.run(HandlerThread.java:61)
```

closed 2019-08-14 by natario1

jianjiankedizai, 2019-09-15:
this still occur 

CameraView version: 2.2.0
Camera engine used: *camera2
Reproducible in official demo app: yes
Device / Android version: xiaomi max2, API 25

2019-09-15 11:48:21.114 1929-2022/com.otaliastudios.cameraview.demo W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-09-15 11:48:21.115 1929-2022/com.otaliastudios.cameraview.demo W/LegacyRequestMapper: convertRequestToMetadata - Ignoring android.lens.focusDistance false, only 0.0f is supported
2019-09-15 11:48:22.358 1929-1955/com.otaliastudios.cameraview.demo I/RequestQueue: Repeating capture request cancelled.
2019-09-15 11:48:22.358 1929-1955/com.otaliastudios.cameraview.demo I/RequestQueue: Repeating capture request set.
2019-09-15 11:48:22.383 1929-2026/com.otaliastudios.cameraview.demo I/Adreno: DequeueBuffer: dequeueBuffer failed
2019-09-15 11:48:22.389 1929-2026/com.otaliastudios.cameraview.demo W/SurfaceTextureRenderer: Surface abandoned, dropping frame. 
    android.hardware.camera2.legacy.LegacyExceptionUtils$BufferQueueAbandonedException
        at android.hardware.camera2.legacy.SurfaceTextureRenderer.swapBuffers(SurfaceTextureRenderer.java:533)
        at android.hardware.camera2.legacy.SurfaceTextureRenderer.drawIntoSurfaces(SurfaceTextureRenderer.java:751)
        at android.hardware.camera2.legacy.GLThreadManager$1.handleMessage(GLThreadManager.java:105)
        at android.os.Handler.dispatchMessage(Handler.java:98)
        at android.os.Looper.loop(Looper.java:163)
        at android.os.HandlerThread.run(HandlerThread.java:61)
2019-09-15 11:48:22.398 1929-2022/com.otaliastudios.cameraview.demo E/RequestQueue: cancel failed: no repeating request exists for request id: 2
2019-09-15 11:48:22.409 1929-2022/com.otaliastudios.cameraview.demo W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
2019-09-15 11:48:22.409 1929-2022/com.otaliastudios.cameraview.demo W/LegacyRequestMapper: Only received metering rectangles with weight 0.
2019-09-15 11:48:22.411 1929-2022/com.otaliastudios.cameraview.demo W/LegacyRequestMapper: convertRequestToMetadata - Ignoring android.lens.focusDistance false, only 0.0f is supported
2019-09-15 11:48:22.413 1929-1934/com.otaliastudios.cameraview.demo I/art: Do partial code cache collection, code=23KB, data=27KB
2019-09-15 11:48:22.413 1929-1934/com.otaliastudios.cameraview.demo I/art: After code cache collection, code=23KB, data=27KB
2019-09-15 11:48:22.413 1929-1934/com.otaliastudios.cameraview.demo I/art: Increasing code cache capacity to 128KB
2019-09-15 11:48:22.480 1929-1955/com.otaliastudios.cameraview.demo W/MediaRecorder: mediarecorder went away with unhandled events
2019-09-15 11:48:22.480 1929-1955/com.otaliastudios.cameraview.demo W/MediaRecorder: mediarecorder went away with unhandled events
2019-09-15 11:48:22.849 1929-1960/com.otaliastudios.cameraview.demo E/AndroidRuntime: FATAL EXCEPTION: LegacyCameraCallback
    Process: com.otaliastudios.cameraview.demo, PID: 1929
    java.lang.NullPointerException: Attempt to invoke virtual method 'android.hardware.camera2.CaptureRequest android.hardware.camera2.impl.CameraDeviceImpl$CaptureCallbackHolder.getRequest(int)' on a null object reference
        at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.onResultReceived(CameraDeviceImpl.java:1792)
        at android.hardware.camera2.legacy.CameraDeviceUserShim$CameraCallbackThread$CallbackHandler.handleMessage(CameraDeviceUserShim.java:316)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:163)
        at android.os.HandlerThread.run(HandlerThread.java:61)
2019-09-15 11:48:22.897 1929-1960/com.otaliastudios.cameraview.demo I/Process: Sending signal. PID: 1929 SIG: 9

--

luangs7, 2020-03-19:
Same error here. Most common with Motorola's devices. 
Using 2.4.0 version

luangs7, 2020-03-19:
@natario1 can you reopen this one please?

PR #551 (2019-08-14): Video bugs
- Fix `takeVideoSnapshot` not working unless you set a max duration (Fixes #553)
- Fix video recording causing crashes on some devices (Fixes #549)