natario1/CameraView #550: Exception #6 on device with only a single camera, front-facing, with demo app

CameraCornet, 2019-08-13:
### Describe the bug
A clear and concise description of what the bug is.

- CameraView version: latest from GitHub
- Reproducible in official demo app: do not know
- Device / Android version: 7.1.1, chromeos / HP chromebook

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
Build and start demo app on device with only a single camera, front-facing

### Expected behavior
Detect absence of BACK camera and on startup default to FRONT.

Change camera switching in the Demo app does not seem to work after a failed start with missing BACK hardware.

### Screenshots
If applicable, add screenshots to help explain your problem.

natario1, 2019-08-13:
Thanks! 
Are you calling `setFacing()`? 
Do you have `app:cameraFacing` set?
Does it work if you remove the two above?

CameraCornet, 2019-08-23:
Again, this is a device with only a single camera, front-only.

The user-interface "switch" front to back fails. Not sure why it gets in a state where switching always fails.

my quick-and-dirty-hack to get the app working on a Front-only device was:

     private Facing mFacing = Facing.FRONT;

Ideally the app would detect failure to open and heal? Or first run of the activity and try to detect devices without front/rear combo?