natario1/CameraView #556: How to get a medium size of video

gabriel-TheCode, 2019-08-16:
### How do I?
I try to record a video with cameraview but I would like it to be between 4 and 6 MB, but the method "cameraVideoSizeSmallest" returns a video of 600kb but unusable because too blurry for facial recognition and the method "cameraVideoSizeSmallest" returns a nice video but too heavy (11Mo)

### Version used
2.0.0-beta04

natario1, 2019-08-19:
You can use `setVideoSize()` and choose a size that you like.

gabriel-TheCode, 2019-08-20:
I think that this method stops the recorder when the desired size is reached, no matter the time of the video.
I want to record a video of 5 seconds with size between 4 and 6 MB, with this method it is possible?

luangs7, 2019-08-31:
Same question here

closed 2019-09-27 by stale[bot]

luangs7, 2019-09-27:
any answer about it?

gabriel-TheCode, 2019-11-05:
Nothing new ...

gabriel-TheCode, 2019-11-22:
I found an alternative by defining the bitRate of my video as I liked according to the resolution I wanted with the method `app:cameraVideoBitRate="@integer/video_bit_rate"`.

 So for a 360p video I set 400kbps. And It works fine !

indracs97, 2019-12-28:
Set max and min bounds to same
app:cameraVideoSizeMinHeight="1920"
app:cameraVideoSizeMinWidth="1080"
app:cameraVideoSizeMaxWidth="1080"
app:cameraVideoSizeMaxHeight="1920"