natario1/CameraView #561: How can i get high Quality Picture?

sunnyj0613, 2019-08-21:
Cameraview Ver 2.1.0 

I want to get high quality picture. 
But i don't know how can i get. 

I just tested using demo. and takePicture() not takePicturSnapshot().

I read docs that takePicture() gets hight Quality Picture and takePictureSanpshot() gets low quality. 

But, Although i used takePicture(), but the result is not high Qulity. 

i will show the sample image. Before take picture, the screen looks good high qulity, but after Taking picture the preview activity show low quality less than before take picture. 

is it possible get to more hight quality picture ?
if that is possible, can you know me?
[image]
[image]

natario1, 2019-08-21:
The demo app resizes the image to 1000 pixels to avoid `OutOfMemory` errors, but your real image should be OK!

sunnyj0613, 2019-08-21:
> The demo app resizes the image to 1000 pixels to avoid `OutOfMemory` errors, but your real image should be OK!

That's work! Thank you so much!!

closed 2019-08-21 by sunnyj0613