natario1/CameraView #569: Frame processor still running after closing camera

najTim, 2019-08-27:
### Describe the bug
Frame processor still running after i close camera and clear all processors. It says that mFrameProcessors size is 0, that there is no frame processor, but i see in log that he is still running. 
 
I tried it on 3 phones, and it was the same, so it`s not phone related problem. I`m closing camera in onPause and reopening it in onResume and attaching frameProcessor there, but it`s not running when i reopen camera, sometimes for few seconds, sometimes longer.

shedeed1, 2019-08-27:
Fotoapparat suffers from the same issue as well, hope it is fixed

najTim, 2019-08-28:
I kinda found the ugly way to do it. You can get id of thread in which frame processor is running, and when u dont need it anymore, just kill it with android.os.Process.killProcess(threadId). 
It`s not good solution, i do not recommend it if you need clean solution. But it`ll get the job done. If someone came up with better solution.

natario1, 2019-08-28:
What do you mean by still running? For how long? How many frames?

I'm OK if we send the very last frames that were taken while the camera was opened, and when they're over, we stop sending. This is intended, you can simply check `camera.isOpened()` in the frame processor method.

If you still believe this is a bug please fill the bug info.

najTim, 2019-08-28:
It`s running for some time. And number of frames varies, from 20 to i don`t know, a lot. Object exists in memory and background thread which it is made in still runs, and is doing some job. Which i don't need, camera is closed. And when i open camera again, frameprocessor does not work, because the first one is still running. 

https://prnt.sc/oylgel

Every D/FACE FOUND: YAAAAAAAAAAY : 1 is a frame which is processed AFTER the camera closed.

natario1, 2019-08-28:
@najTim your logs are likely showing up after **your own** asynchronous processing. You should add a log line in `onCameraClosed` and one at the beginning of `process` method. Then count how many process calls you get **after** onCameraClosed.

najTim, 2019-08-28:
https://prnt.sc/oynue7
https://prnt.sc/oynuls
https://prnt.sc/oynuzb

natario1, 2019-08-28:
@najTim can you move the counter++ and the counter log to the beginning of the the `process` method which is the part that's related to this library?

najTim, 2019-08-29:
@natario1 Ye, it kinda does stop. My bad. The problem is with list of Tasks which get frames. I`ll work it out with Google and i'm gonna post answer here and close the discussion when i get the answer so that next who have problem with this, can find the answer.

natario1, 2019-08-29:
@najTim see https://natario1.github.io/CameraView/docs/frame-processing.html . You should use `Tasks.await()` or a `CountDownLatch`

najTim, 2019-08-29:
> @najTim see https://natario1.github.io/CameraView/docs/frame-processing.html . You should use `Tasks.await()` or a `CountDownLatch`

Yes, this is the answer. I should have read documentation more carefully. Thank you mate. Keep the great work. Cheers.

closed 2019-08-29 by najTim

alexei-28, 2020-02-25:
In my activity A I add frame processor like this:

```
 implementation "com.otaliastudios:cameraview:2.6.1"

 dataBinding.cameraView.addFrameProcessor(object : FrameProcessor {          
            override fun process(frame: Frame) {
                Debug.d(TAG, "process_start: call_doRecognizeCheck()")
                scanCheckViewModel.doRecognizeCheck(frame)
            }
        })
```

Nice. It's work fine.
But I need , after forward to another activity B to stop frame processor. Is it possible?