natario1/CameraView #575: setPictureSize not working always

umitemre, 2019-09-01:
Hello,

I want to set both preview size and picture size the same dimensions. I couldn't find any related function to do this. So I tried to get preview size from setPreviewStreamSize and call setPictureSize right after but it gets called never. After I go to another activity and back to camera activity then it calls setPictureSize callback.

I don't know what's wrong but somethings definetly wrong here.

Can you help me about the issue?

natario1, 2019-09-04:
I will consider this! For now you should change the picture size, then call close(), then call open().

umitemre, 2019-09-05:
> I will consider this! For now you should change the picture size, then call close(), then call open().

While you are considering, can you also consider making onCameraReady event too? It can inform when camera is ready to open or ready to spit out previews and can be last event call before onCameraOpened

closed 2019-10-02 by natario1