natario1/CameraView #577: WARNING: API 'variant.getJavaCompile()'

unengchen, 2019-09-04:
WARNING: API 'variant.getJavaCompile()' is obsolete and has been replaced with 'variant.getJavaCompileProvider()'.
It will be removed at the end of 2019.
For more information, see https://d.android.com/r/tools/task-configuration-avoidance.
To determine what is calling variant.getJavaCompile(), use -Pandroid.debug.obsoleteApi=true on the command line to display more information.
Affected Modules: cameraview

natario1, 2019-09-11:
You must be using a newer gradle version than this project does. When this warning appears in this project, we'll fix it.

closed 2019-09-11 by natario1

Johnson145, 2019-10-08:
@unengchen @natario1 

[Here](https://github.com/Johnson145/CameraView/commit/7a412ab11f6ebbae859b930e3f080542227a83ff) is a fix for this error.

(Not raising a PR, because my fork has diverged too much).