natario1/CameraView #578: Exception occurs while taking a video on Xiaomi MI A1, Android Pie

zauli83, 2019-09-04:
Cannot take video, exception occurs
- CameraView version: 2.2.0
- Camera engine used: both
- Reproducible in official demo app: no
- Device / Android version: Xiaomi Mi A1, Android Pie

### To Reproduce
1) Take a video

### XML layout
```
<?xml version="1.0" encoding="utf-8"?>
<androidx.constraintlayout.widget.ConstraintLayout xmlns:android="http://schemas.android.com/apk/res/android"
        xmlns:app="http://schemas.android.com/apk/res-auto"
        xmlns:tools="http://schemas.android.com/tools"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        tools:context=".record.RecordVideoActivity">

    <com.otaliastudios.cameraview.CameraView
            android:id="@+id/camera"
            android:layout_width="0dp"
            android:layout_height="0dp"
            android:keepScreenOn="true"
            app:cameraGestureTap="none"
            app:cameraGestureLongTap="none"
            app:cameraGesturePinch="zoom"
            app:cameraGestureScrollHorizontal="none"
            app:cameraGestureScrollVertical="none"
            app:cameraFacing="back"
            app:cameraMode="video"
            app:cameraAudio="on"
            app:cameraPlaySounds="true"
            app:cameraUseDeviceOrientation="true"
            app:layout_constraintTop_toTopOf="parent"
            app:layout_constraintStart_toStartOf="parent"
            app:layout_constraintEnd_toEndOf="parent"
            app:layout_constraintBottom_toBottomOf="parent">

    </com.otaliastudios.cameraview.CameraView>

    <Button
            android:id="@+id/start"
            android:layout_width="wrap_content"
            android:layout_height="wrap_content"
            android:text="START"
            app:layout_constraintBottom_toBottomOf="parent"
            app:layout_constraintStart_toStartOf="parent" />

    <Button
            android:id="@+id/stop"
            android:layout_width="wrap_content"
            android:layout_height="wrap_content"
            android:text="STOP"
            app:layout_constraintBottom_toBottomOf="parent"
            app:layout_constraintEnd_toEndOf="parent" />

</androidx.constraintlayout.widget.ConstraintLayout>

```

Code:

```
class RecordVideoActivity : AppCompatActivity() {

    override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)
        setContentView(R.layout.activity_record_video)
        CameraLogger.setLogLevel(CameraLogger.LEVEL_VERBOSE)
[... 120 lines cut ...]
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: start failed.
        at com.otaliastudios.cameraview.engine.CameraEngine.onVideoResult(CameraEngine.java:1211)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onVideoResult(Camera1Engine.java:379)
        at com.otaliastudios.cameraview.video.VideoRecorder.dispatchResult(VideoRecorder.java:93)
        at com.otaliastudios.cameraview.video.Full1VideoRecorder.dispatchResult(Full1VideoRecorder.java:51)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStop(FullVideoRecorder.java:206)
        at com.otaliastudios.cameraview.video.VideoRecorder.stop(VideoRecorder.java:70)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:183)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:62)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:339)
        at com.otaliastudios.cameraview.engine.CameraEngine$23.run(CameraEngine.java:1151)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)
     Caused by: java.lang.RuntimeException: start failed.
        at android.media.MediaRecorder.start(Native Method)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:177)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:62) 
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:339) 
        at com.otaliastudios.cameraview.engine.CameraEngine$23.run(CameraEngine.java:1151) 
        at android.os.Handler.handleCallback(Handler.java:873) 
        at android.os.Handler.dispatchMessage(Handler.java:99) 
        at android.os.Looper.loop(Looper.java:193) 
        at android.os.HandlerThread.run(HandlerThread.java:65)

natario1, 2019-09-04:
You might be missing storage permissions. Try defining your output File like the demo app does.

zauli83, 2019-09-05:
It doesn't work. I've defined the output video as  File(filesDir, "video.mp4") ( as the demo does), i've also checked the permissions, but doesn't work. Same exception here.

natario1, 2019-09-11:
But you say that this does not happen in the demo app. So I recommend trying to spot the difference that makes this crash in your app.

closed 2019-10-08 by stale[bot]